ferrandi/PandA-bambu #4: Fail to configure

jackfrye, 2017-12-29:
Version: 0.9.5
Environment: Ubuntu 16.04.3 

Steps to replicate:
1. Install packages
$ sudo apt-get install autoconf autoconf-archive automake libtool g++ gcc-4.8 g++-4.8 gcc-4.9 g++-4.9 gcc-5 g++-5 gcc-4.8-plugin-dev gcc-4.9-plugin-dev gcc-5-plugin-dev  gcc-4.8-multilib gcc-4.9-multilib gcc-5-multilib libboost-all-dev libmpc-dev libmpfr-dev libxml2-dev liblzma-dev libmpfi-dev zlib1g-dev libicu-dev bison doxygen flex graphviz iverilog verilator make libsuitesparse-dev libglpk-dev

2.
Download the package from Firefox and untar

3. Create directories
 $mkdir obj
 $cd obj

4. Configure without Mentor License
../configure --prefix=/opt/panda --enable-flopoco --enable-icarus --enable-verilator --enable-xilinx --enable-modelsim --enable-altera --enable-opt --enable-lattice --enable-release

5. Attempt to make
$ make
Make will say there are no targets

I assume this is a problem with the configure script not generating proper makefile

Repeating step 4 without any add-ons appears to work
4. ../configure --prefix=/opt/panda 
5. make

Documentation or configurations script should be updated. Are all these proprietary pieces of software like Xilinx's tools required to be installed for the configure script to configure properly when the --enable-<proprietary tool> option is enabled?

lattuada, 2017-12-31:
Yes, --enable-<tool> requires that the tool is already installed since it enables the wrapper for the simulation/synthesis tool. If the configure does not find the tool, it aborts.
We will update the INSTALL instructions to clarify this.

jackfrye, 2018-01-01:
The tools did install properly without the --enables. I will close.

closed 2018-01-01 by jackfrye