ferrandi/PandA-bambu #14: HOW TO GENERATE VCD FILE

lewiz-sunny, 2020-08-14:
Hello,
I am trying to convert simple counter code from C to verilog, but somehow my simulation shows it only runs for 1 cycle. Also, I am looking for vcd file. Is there any way to generate a .vcd file for Vivado simulator?

my C program is as follows.
#include <stdio.h>

int main() 
{
   int i, start,end, count;
   start = 0;
   end = 10;
   count = 0;
   for(i = start; i <= end; i++) 
      count = count + 1;

   //return count;
}

and I am using bambu count.c --simulate --simulator=XSIM command to generate the verilog file.

Thank you.

chrpilat, 2020-08-14:
Hello,
did you check the output Verilog?
I am afraid that, without a visible output, the compiler optimizes out the entire code.

Try to set the main with an integer return value and return the counter.

thanks,
Christian

Sent from my mobile. Please excuse the brevity, spelling and punctuation.

lewiz-sunny, 2020-08-14:
Thank you Christian for your reply.
When I do return count in main program, it gives me following error:
error -> Error in generating the expected test results.

thanks,
Sunny

chrpilat, 2020-08-14:
Otherwise, you can put your code inside a function instead of the main one. The tool will automatically create a wrapper to create the expected results.

Thanks,
Christian 

Sent from my mobile. Please excuse the brevity, spelling and punctuation.

ferrandi, 2020-08-14:
In case the main returns a value different than zero the simulator raises an error. This is exploited during the bambu regressions.
One solution is to put your code in a function not named main or pass this option to bambu --accept-nonzero-return.
Cheers,
Fabrizio

ferrandi, 2020-08-14:
One thing that should help in understanding what bambu does. Passing --print-dot to bambu will expose the optimizations done on your code. Under directory HLS_output/dot/ you will find many dot files describing the IR on which bambu is working.
Finally, to generate a vcd file you just need to add --generate-vcd to the bambu options.
cheers,
Fabrizio

lewiz-sunny, 2020-08-14:
Thank you Christian and Ferrandi,
The --accept-nonzero-return option did gave me the return port value as 1000, but somehow it still mismatches with the expected value and shows error in simulation.

ferrandi, 2020-08-14:
I tried to reproduce the issue but I miss some information. 
Which is the version of bambu are you using. Is the one from git master or from the panda-0.9.7-dev branch?
Moreover, which GCC/Clang compiler are you using? Could you please check the list of compilers your bambu can use?
bambu -help reports even the list of compilers working with your environment. Something like:
    --compiler=<compiler_version>
        Specify which compiler is used.
        Possible values for <compiler_version> are:
            I386_GCC48
            I386_GCC49
            I386_GCC5
            I386_GCC6
            I386_GCC7
            I386_GCC8
            I386_CLANG4
            I386_CLANG5
            I386_CLANG6
            I386_CLANG7
            I386_CLANG8
            I386_CLANG9

Finally, is the code you used the same you put in the first post?

Cheers,

Fabrizio

PS which VIVADO are you using? I tested XSIM with Vivado 2019.2 and it works.

lewiz-sunny, 2020-08-14:
Version: PandA 0.9.6 - Revision 5e5e306b86383a7d85274d64977a3d71fdcff4fe
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Vivado 2020.1
and 

#include <stdio.h>
int main() 
{
   int i,count,j;
   count = 0;
   i = 0;
   j = 1;
   while(i < 1000)
   {
      count = count + j;
      i++;
	}
   return count;
}

ferrandi, 2020-08-14:
The expected value computed by the C code is 1000%256, which is 232. This happens since on most Unix-based systems you have only 8-bit return values.
So, if you rename main in test everything should work as expected.
Cheers,
Fabrizio

lewiz-sunny, 2020-08-14:
Thank you very much Ferrandi. I really appreciate that. I can see it working now.
Just for curiosity, does panda also produces a Verilog code which is human readable? I am going through the top.v but I am not sure what it is doing exactly.

Thanks,
Sunny

ferrandi, 2020-08-14:
Sunny,
The Verilog is mainly a structural code following the FSM+Datapath model. The example you used is quite simple since the compiler is able to understand that the returning value is always 1000. So, the Datapath is having just a few components meant for returning the 1000 constant.
 In more complex cases, I suggest starting from the FSM associated with each function. For example, if the function is named test and you passed --print-dot, you should look to the file HLS_output/dot/test/HLS_STGraph.dot that can be graphically viewed with xdot. There is a simpler version with some information missing named HLS_output/dot/test/fsm.dot. From the FSM and from the C code included in the states you should be able to understand what is done by the Datapath.
Other tools produce mainly a behavioral Verilog while bambu generates structural code. I tend to consider this more controllable/maintainable than the behavioral code. Understanding how many muxes/adders/multipliers are intended is straightforward for example when you look to the code.
Cheers,
Fabrizio

lewiz-sunny, 2020-08-14:
Thank you for the explanation.
Closing this issue.

Cheers,
Sunny

closed 2020-08-14 by lewiz-sunny