ferrandi/PandA-bambu #15: Cannot allocate the memory.

lewiz-sunny, 2020-08-26:
Hi, 
This is not an issue, but I am having difficulty allocating memory. I have a function in C which uses calloc function as shown below:
void *chk_alloc
( unsigned n,		/* Number of elements */
  unsigned size		/* Size of each element */
)
{ 
  void *p;
  p = calloc(n,size);
  if (p==0)
  { //fprintf(stderr,"Ran out of memory (while trying to allocate %d bytes)\n",
    //  n*size);
    exit(1);
  }
return p;
}

Is there any workaroud this problem? How can I allocate memory in Hardware without using Calloc function. 
Any suggestions will be very helpful. 

Thank you.

ferrandi, 2020-08-26:
Hi,
the dynamic memory allocation is managed by this C function synthesized by bambu.
[MemMGR](https://github.com/ferrandi/PandA-bambu/blob/master/etc/libbambu/memmgr.c)
changing the define you may increase the static pool used by the memory allocator.

_"How can I allocate memory in Hardware without using Calloc function."_
Declaring a standard static array? Is this what you need?
Cheers,
Fabrizio

lewiz-sunny, 2020-08-26:
I think I was able to generate the verilog code but failed in simulation:

I am getting the following error for XSIM simulator:
ERROR: [VRFC 10-536] constant expression required

ferrandi, 2020-08-26:
I’ll check what XSIM says.
In the meantime, have you tried Verilator as verilog simulator?
Cheers,
Fabrizio

lewiz-sunny, 2020-08-26:
Yes, and I can see some waveforms through generate-vcd option. I guess its working with Verilator. I am actually trying to synthesize matrix multiplier, adder and transpose functions written in C to verilog.

ferrandi, 2020-08-26:
Verification with bambu is done by co-simulation, that is, by executing C and the Verilog and comparing the results. There are different ways to pass stimulus. The simplest is with option —generate-tb=comma separated pairs of arguments=value. Arrays could be passed as well but the value should be enclosed in double quotes and following the C syntax. As example —generate-tb=A=“{1,2,3,4}”,B=“{{1,2},{1,2}}” is specifying the stimulus for the top function where A and B are the names of the top function parameters and A is a two elements 1D array while B is a 2x2 array. The tutorials should help in this.

lewiz-sunny, 2020-08-26:
Thank you very much for providing this. I will try to do the co-simulation and will get back to you with the status. 

Thank you once again.

Cheers,
Sunny

lewiz-sunny, 2020-09-02:
Hi @ferrandi 
I am trying to do simple sorting program now. I am getting following error while trying to simulate. 

error -> unexpected condition (ic != nullptr)

The code for array sorting is as follows:
// An iterative implementation of quick sort 
#include <stdio.h> 

// A utility function to swap two elements 
void swap(int* a, int* b) 
{ 
	int t = *a; 
	*a = *b; 
	*b = t; 
} 

/* This function is same in both iterative and recursive*/
int partition(int arr[], int l, int h) 
{ 
	int x = arr[h]; 
	int i = (l - 1); 
    
	for (int j = l; j <= h - 1; j++) { 
		if (arr[j] <= x) { 
			i++; 
			swap(&arr[i], &arr[j]); 
		} 
	} 
	swap(&arr[i + 1], &arr[h]); 
	return (i + 1); 
} 

/* A[] --> Array to be sorted, 
l --> Starting index, 
h --> Ending index */
void quickSortIterative(int arr[], int l, int h) 
{ 
	// Create an auxiliary stack 
	int stack[h - l + 1]; 
    
	// initialize top of stack 
	int top = -1; 
    
	// push initial values of l and h to stack 
	stack[++top] = l; 
	stack[++top] = h; 
    
	// Keep popping from stack while is not empty 
	while (top >= 0) { 
		// Pop h and l 
		h = stack[top--]; 
		l = stack[top--]; 
        
		// Set pivot element at its correct position 
		// in sorted array 
		int p = partition(arr, l, h); 
        
		// If there are elements on left side of pivot, 
		// then push left side to stack 
		if (p - 1 > l) { 
			stack[++top] = l; 
			stack[++top] = p - 1; 
		} 
        
		// If there are elements on right side of pivot, 
		// then push right side to stack 
		if (p + 1 < h) { 
			stack[++top] = p + 1; 
			stack[++top] = h; 
		} 
	}
} 

// A utility function to print contents of arr 
void printArr(int arr[], int n) 
{ 
	int i; 
	for (i = 0; i < n; ++i) 
		printf("%d ", arr[i]); 
	 printf("\n ");
}  

// Driver program to test above functions 
int main() 
{ 
	int arr[] = { 4, 3, 5, 2, 1, 3, 2, 3 }; 
	int n = sizeof(arr) / sizeof(*arr); 
	quickSortIterative(arr, 0, n - 1); 
	printArr(arr, n); 
	return 0; 
}

ferrandi, 2020-09-05:
Thanks for reporting this issue.
The problem is related to a variable dimension array declaration.
I fixed the issue with GCC but I have not yet fixed it with Clang. Consider that this is going to require a dynamic memory allocation and so in terms of cycles you will have worst latency than declaring a fixed-length array.
I'll update as soon as I completely fix the issue.
Fabrizio

ferrandi, 2020-09-11:
GCC support for variable length array has been added.
Check this https://github.com/ferrandi/PandA-bambu/tree/panda-0.9.7-dev
CLANG requires a lowering step not yet implemented.

lewiz-sunny, 2020-09-11:
Hi @ferrandi ,
Thank you for providing the updates so quickly. I will try to use it more and let you know if I come across any more issues. 

Cheers,
Sunny

lewiz-sunny, 2020-09-17:
Hi @ferrandi ,
I tried installing the panda 0.9.7 version and running the same code but it still shows the same error. Am I doing something wrong? I also tried to use the --compile=I386_GCC8 option but still no change. 

Thanks,
Sunny

lewiz-sunny, 2020-09-17:
Hi @ferrandi 
I tried reinstalling the panda with 0.9.7 version, but I am still facing the same issue. Am I doing something wrong? 

Thank you,
Sunny

ferrandi, 2020-09-18:
Hi Sunny,

I tested against GCC 4.8, 4.9, 5, 6, 7, and 8, and on my side, it works.
Could you please share the output log of bambu? On which Linux distribution are you working?
Which options have you passed to the panda configure?
Cheers,
Fabrizio

lewiz-sunny, 2020-09-18:
bambu ../quick_sort_iterative.c --simulate --simulator=XSIM --generate-vcd --compiler=I386_GCC7
 ==  Bambu executed with: bambu --simulate --simulator=XSIM --generate-vcd --compiler=I386_GCC7 ../quick_sort_iterative.c 

 Version: PandA 0.9.6 - Revision 5e5e306b86383a7d85274d64977a3d71fdcff4fe-master

Target technology = FPGA
error -> unexpected condition (ic != nullptr)

I am running on Ubuntu 18.04 LTS (5.4.0-47-generic Linux kernel version).
While configuring Panda I used following options.
../configure --prefix=/opt/panda --enable-verilator --enable-xilinx --enable-opt 

My current GCC version is 8.4.0

ferrandi, 2020-09-19:
Hi Sunny,
I wasn't clear. The patch is on a branch and it is not yet in master.
please change the branch with
`git checkout panda-0.9.7-dev`
then restart the building process from make -f Makefile.init, followed by the configure.
Cheers,
Fabrizio

lewiz-sunny, 2020-09-21:
hi @ferrandi 
The following error pops up while doing the make
../../../../src/HLS/simulation/CTestbenchExecution.cpp:46:10: fatal error: config_HAVE_I386_CLANG8_COMPILER.hpp: No such file or directory
 #include "config_HAVE_I386_CLANG8_COMPILER.hpp"

ferrandi, 2020-09-21:
Hi Sunny,
You need to re-run the configure from the command line.
Cheers,
Fabrizio

lewiz-sunny, 2020-09-21:
Steps I did:
2046  git checkout panda-0.9.7-dev
2048  make -f Makefile.init
 2049  cd obj/
 2050  history | grep configure
 2051  ../configure --prefix=/opt/panda --enable-verilator --enable-xilinx --enable-opt
 2052  make

ferrandi, 2020-09-21:
Could you please repeat command 2051.
It case it is not working, please remove directory obj.
Usually re-running the configure it works.
Cheers,
Fabrizio

lewiz-sunny, 2020-09-23:
Hi @ferrandi 
I was able to convert quick sort program now. Thank you for your prompt response and help. I really appreciate it.

Thank you,
Sunny

closed 2020-11-06 by ferrandi