ferrandi/PandA-bambu #17: Unexpected behavior of the synthesized RTL model (vmul)

askamkin, 2020-10-06:
Hello!

I am trying to synthesize an RTL implementation of the following function (just to get started w/ Bambu):

```c
#include <stddef.h>

float vmul(float *a, float *b, size_t n) {
  float c = 0.0;
  for (size_t i = 0; i < n; i++) {
    c += a[i] * b[i];
  }
  return c;
}
```

The command line is as follows:
```
bambu --top-fname vmul --device-name=5CSEMA5F31C6 vmul.c
```
```
Version: PandA 0.9.6 - Revision 5e5e306b86383a7d85274d64977a3d71fdcff4fe
GCC: gcc version 7.5.0
```

It seems that the generated Verilog code does not work properly (or I am using the tool in an incorrect way).

Here are two particular cases.

- If n == 1 and the memory returns 1.0/1.0 (3f800000 3f800000), the result is 0.0 (00000000), while 1.0 (3f800000) is expected.
- If n == 2 and the memory returns 1.0/1.0 for two times, the result is 1.0 (3f800000), while 2.0 (40000000) is expected.

[image]

[image]

It looks like the RTL performs one iteration less than required.

Another issue is that the operation arguments (a, b, and n) should be set until done_port appears (otherwise, the model does not work). Is it really so?

ferrandi, 2020-10-06:
Hello Alexander,
it looks that you do not share the test.xml used to test very the module. In case you left the creation to bambu, the test.xml is not correct. It is not easy to automatically generate the test.xml when pointers are involved.
Could you please check if this is going to work?
bambu --top-fname=vmul --device-name=5CSEMA5F31C6 vmul.c --generate-tb=a="1,2",b="3,4",n="2"
In case you are going to use bambu 0.9.7 or above the generate-tb syntax change slightly in
bambu --top-fname=vmul --device-name=5CSEMA5F31C6 vmul.c --generate-tb=a="{1,2}",b="{3,4}",n="2"

> Another issue is that the operation arguments (a, b, and n) should be set until done_port appears (otherwise, the model does not work). Is it really so?

The default expects that arguments are stable along the computation. Passing -registered-inputs=yes changes the default and so the inputs are not required to be stable along the computation anymore.

cheers,
Fabrizio

askamkin, 2020-10-06:
Fabrizio, thanks for the quick reply!

> Could you please check if this is going to work?
> bambu --top-fname=vmul --device-name=5CSEMA5F31C6 vmul.c --generate-tb=a="1,2",b="3,4",n="2"

Something like that:

```
Total number of flip-flops in function vmul: 159
syntax error at line number |1|	text is |,|
Parse Error
Please report bugs to <panda-info@polimi.it>
```

Best wishes,
Alexander

ferrandi, 2020-10-06:
Add the graph parentheses as in 0.9.7 version

askamkin, 2020-10-06:
It's better.

askamkin, 2020-10-06:
Fabrizio,

As far as I understand, the RTL model assumes that a response from the memory goes exactly one cycle later after the request. M_DataRdy is not in use. Right?

In my example, the model took zero data in the beginning (due to the delay), which led to the incorrect result.

Best wishes,
Alexander

ferrandi, 2020-10-07:
Alexander,
The default used by bambu is --experimental-setup=BAMBU-BALANCED-MP.
In this case the assumption is that any reference to memory works as it is referring to a BRAM. This is controlled by this option
--memory-allocation-policy=ALL_BRAM passed when the experimental setup is BAMBU-BALANCED-MP.
Memory handshaking protocol is used when you pass to bambu this option: --memory-allocation-policy=NO_BRAM overriding the default you have with BAMBU-BALANCED-MP experimental setup.
Cheers,
Fabrizio

askamkin, 2020-10-07:
Many thanks, Fabrizio!

closed 2020-11-06 by ferrandi