ferrandi/PandA-bambu #18: Increasing the memory bandwidth and parallel data processing

askamkin, 2020-10-12:
I could not find information on how to customize the memory architecture in Bambu. In particular, I am interested in how to increase the data line width so as to be able to load large chunks of data at a cycle and process them in parallel.

For example, I need a parallel implementation of the following function for multiplying vectors of length 64:

```c
#define N 64

float vmul(float *a, float *b) {
  float c = 0.0;
  for (size_t i = 0; i < N; i++) {
    c += a[i] * b[i];
  }
  return c;
}
```

I would like to load two vectors at a cycle and calculate their scalar multiplication in parallel (w/ parallel reduction). Is it possible?

ferrandi, 2020-10-12:
In bambu there is partial support for vectorized instructions. Unfortunately, the vectorized floating-point instructions are not yet been added to the Bambu IPs library. If you change in you code the type float into unsigned you will see an accelerator reading two chunks of 128 bits at each cycle from the BRAMs.
Other forms of parallel reductions are not yet been implemented.
Cheers,
Fabrizio

ferrandi, 2020-10-12:
To activate the vectorization pass this option: -fopenmp. Vectorization currently works only for GCC backend.
Cheers,
Fabrizio

askamkin, 2020-10-12:
Thank you,  Fabrizio.

closed 2020-11-06 by ferrandi