ferrandi/PandA-bambu #19: cannot setup bambu

joker97max, 2020-10-26:
before 2nd step for install Bambu i see that:

~/Projects/panda-0.9.6/obj$ ../configure --prefix=/opt/panda --enable-flopoco --enable-icarus --enable-verilator --enable-xilinx --enable-modelsim --enable-altera --enable-opt --enable-lattice --enable-release --with-mentor-license=license-string
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for gawk... (cached) mawk
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... -lfl
checking whether yytext is a pointer... yes
checking for bison... bison -y
checking for g77... no
checking for xlf... no
checking for f77... no
checking for frt... no
checking for pgf77... no
checking for cf77... no
checking for fort77... no
checking for fl32... no
checking for af77... no
checking for xlf90... no
checking for f90... no
checking for pgf90... no
checking for pghpf... no
checking for epcf90... no
checking for gfortran... no
checking for g95... no
checking for xlf95... no
checking for f95... no
checking for fort... no
checking for ifort... no
checking for ifc... no
checking for efc... no
checking for pgfortran... no
checking for pgf95... no
checking for lf95... no
checking for ftn... no
checking for nagfor... no
checking whether we are using the GNU Fortran 77 compiler... no
checking whether  accepts -g... no
checking if g++ supports C++17 features without additional flags... no
[... 134 lines cut ...]
checking whether the Boost::System library is available... yes
checking for main in -lboost_system... yes
checking whether the Boost::Date_Time library is available... yes
checking for exit in -lboost_date_time... yes
checking whether the Boost::Filesystem library is available... yes
checking for main in -lboost_filesystem... yes
checking whether the Boost::Regex library is available... yes
checking for exit in -lboost_regex... yes
checking if -DBOOST_NO_SCOPED_ENUMS or BOOST_NO_CXX11_SCOPED_ENUMS is required... no
checking if -rpath is fine... yes
checking linux distribution... ubuntu_20.04
checking if directory is a git repository... no
checking if directory is a git worktree... no
checking bdd.h usability... yes
checking bdd.h presence... yes
checking for bdd.h... yes
checking fdd.h usability... yes
checking fdd.h presence... yes
checking for fdd.h... yes
checking bvec.h usability... yes
checking bvec.h presence... yes
checking for bvec.h... yes
checking if ALTERA quartus_sh is present in /opt/altera/*... no
checking if ALTERA quartus_sh is present in /opt/intelFPGA/*... no
configure: error: ALTERA settings script required

joker97max, 2020-10-26:
this libs i'm install

- autoconf
- autoconf-archive
- automake
- libtool
- g++ from 7 to 10
- gcc- from 7 to 10
- gcc-like as-plugin-dev
- gcc-like as-multilib
- g++-like as-multilib
- gfortran-like as
- clang-like as
- libclang-like as-dev
- libbdd-dev
- libboost-all-dev
- libmpc-dev
- libmpfr-dev
- libxml2-dev
- liblzma-dev
- libmpfi-dev
- zlib1g-dev
- libicu-dev
- bison
- doxygen
- flex
- graphviz
- iverilog
- verilator
- pkg-config
- libsuitesparse-dev
- libglpk-dev
- make

ferrandi, 2020-10-26:
Dear Max,
from the options you passed to the configure, it seems that you have just copied the options suggested by the INSTALL file. This way is possible only in case you have all the tools installed on your system.
For example, passing --enable-altera requires to have Intel/Altera tools installed. In case you do not have such a tool, please remove such option. Same thing for -enable-xilinx --enable-modelsim --enable-altera --enable-lattice --with-mentor-license=license-string.
Cheers,
Fabrizio

joker97max, 2020-10-26:
Okey, thanks for answer, Fabrizio.

ferrandi, 2020-11-03:
I'm going to close since it seems that it is just a misinterpretation of the INSTALL instructions.

closed 2020-11-03 by ferrandi