ferrandi/PandA-bambu #20: Setting the target language version/standard

askamkin, 2020-11-05:
Is it possible to force Bambu to follow IEEE 1364-2005 (and not to use SystemVerilog constructs)?

This would facilitate using Bambu w/ open-source EDA tools.

For example, `BRAM_MEMORY_NN_SV_CORE` includes the following line:
```verilog
reg [(BRAM_BITSIZE/8)-1:0][7:0] memory_a [0:n_mem_elements-1];
```

which causes Icarus Verilog to report a syntax error.

ferrandi, 2020-11-05:
bambu supports different devices. SystemVerilog for memory constructs is required by Quartus from Altera/Intel.
There are different styles for different target devices. Have you tried the default targeting a zynq device?
Have you checked bambu with this option --device-name=xc7z020-1clg484-YOSYS-VVD? This last option targets a zynq but the flow uses both yosys and Vivado place and route flow.

ferrandi, 2020-11-05:
One additional note. BRAM_MEMORY_NN_SV_CORE was written in SystemVerilog since this is the only way I've found to infer a BRAM with byte-enabling when Altera/Intel devices are considered.

ferrandi, 2020-11-05:
"This would facilitate using Bambu w/ open-source EDA tools."
Icarus may have some problems but Verilator supports such a Verilog without any problem.

askamkin, 2020-11-05:
Thanks for quick response!
> Have you checked bambu with this option --device-name=xc7z020-1clg484-YOSYS-VVD?
Not yet. I'll try tomorrow.
> Icarus may have some problems but Verilator supports such a Verilog without any problem.
Sure. The thing is that we are using a VPI-based testbench development tool, and Verilator is not very friendly with VPI.

askamkin, 2020-11-06:
Now it works. Thank you!

closed 2020-11-06 by ferrandi