ferrandi/PandA-bambu #21: Failure: error -> -m parameter not supported: -m64

noahsherrill, 2020-11-05:
I see the error `error -> -m parameter not supported: -m64 `  when running in an environment in which `HAVE_I386_CLANG7_M32` and `HAVE_I386_CLANG7_MX32` are 0, but `HAVE_I386_CLANG7_M64` is 1. The `OPT_gcc_m32_mx32` is set here: https://github.com/ferrandi/PandA-bambu/blob/5e5e306b86383a7d85274d64977a3d71fdcff4fe/src/BambuParameter.cpp#L3748 All of the parameter strings contain a space at the end. The option is read here: https://github.com/ferrandi/PandA-bambu/blob/5e5e306b86383a7d85274d64977a3d71fdcff4fe/src/wrapper/treegcc/gcc_wrapper.cpp#L3130 However, a space is only anticipated for one case. For the other cases, it looks like it will always trigger an error because it does not expect a space in the parameter string.

ferrandi, 2020-11-06:
Thank you very much for the bug report. I've internally patched the code and now I'm waiting for the regression to end.
Once the regression complete I'll publicly push the patch.
Cheers,
Fabrizio

ferrandi, 2020-11-08:
@noahsherrill, is the above patch fixing the issue?

noahsherrill, 2020-11-10:
Yes, that fixed the issue. Thank you!

ferrandi, 2020-11-11:
Ok, I’m going to close the issue. Thank you very much for raising the issue.

closed 2020-11-11 by ferrandi