ferrandi/PandA-bambu #24: Synthesizing neural networks with bambu

mrlebedev, 2020-12-23:
Hello,
I am trying to synthesize a simple neural network using your workflow: onnx-tvm-llvm-bambu. I've managed to create a llvm model of the network, but when I synthesize it with bambu I get a segfault when 

> Running pass 'clang7_plugin_dumpGimpleSSA' on module 'model.ll'

The same issue arised when I tried the neural network examples from examples/onnx, except 00_lenet and i3_vgg16 that failed because of

> assignCodeAuto kind not supported: InsertElement
Plugin Error
UNREACHABLE executed at ../../../etc/clang_plugin/dumpGimple.cpp:486!

I've noticed that these examples are not included into examples/onnx benchmark.
Are there any limitations on neural networks synthesis? Should there be some wrapper code like in the other examples? Or maybe I am using the wrong compiler/options/etc.? 

I used the following command line:

> bambu model.ll -fno-inline -fno-inline-functions -v4 --compiler=I386_CLANG7

Thank you.

ferrandi, 2020-12-24:
Hi Mikhail,
The clang plugin we use does not support vectorized code. InsertElement is usually related to vectorized code. So, you -scalarize your .ll with opt, and then the clang plugin should be facilitated to get the work done.
Sharing the .ll would help anyway to understand what is going to happen with segfault you faced. Are you using the panda-0.9.7-dev branch?

`bambu model.ll -fno-inline -fno-inline-functions -v4 --compiler=I386_CLANG7`
Note that -fno-inline -fno-inline-functions are actually relevant when you pass C/C++ functions. When you start from a .ll you are on your own for all the Clang/llvm compiler optimizations. Bambu does many optimizations but these are not under the control of the -f option.
cheers,
Fabrizio

mrlebedev, 2020-12-24:
Thank you, Fabrizio!
Scalarization with -scalarizer -scalarize-load-store really helped and bambu managed to generate the Verilog code. I think that my problem with segfault is now irrelevant so I can proceed with my experiments.
I used the master branch of panda.

Merry Christmas, by the way!
Mikhail

ferrandi, 2020-12-24:
please use panda-0.9.7-dev. It includes many fixes and it is quite stable.
Merry Christmas!
Fabrizio

closed 2021-02-10 by ferrandi