ferrandi/PandA-bambu #26: Literals not getting expanded properly

ymherklotz, 2021-01-15:
Hi,

It seems that literals in Bambu do not get expanded with the correct sign. Considering the following example:

```C
int main() {
  return 0xffffffffL == -1;
}
```

When run on GCC/Clang, this returns 0, because `0xffffffffL` should become `0x00000000ffffffff`, which is not equal to -1. Instead, Bambu seems to return 1 for the example, meaning the literal got expanded to `(long)(-1)`.

## Versions and commands run

I have attached the testcase [here: testcase2.zip](https://github.com/ferrandi/PandA-bambu/files/5820693/testcase2.zip).  This seems to occur with the following bambu version:

```
Version: PandA 0.9.7-dev - Revision 6f2d6eb92cb1507af12248a503ef39990d3d9c23-SROA_TCAD_release
```

And the following commands were run to reproduce this behaviour:

```sh
bambu test.c >bambu.log 2>&1
iverilog top.v tb.v -o top
./top >out.iverilog.txt

gcc test.c -o test
./test
echo "checksum = 0000000$?" >out.gcc.txt

cat out.iverilog.txt | tr 'A-Z' 'a-z' >out.iverilog.lw.txt
cat out.gcc.txt | tr 'A-Z' 'a-z' >out.gcc.lw.txt

diff out.iverilog.lw.txt out.gcc.lw.txt
```

With the following testbench in Verilog:

```verilog
module testbench;
   reg clock, reset, start_port;
   wire done_port;
   wire [31:0] return_port;

   main m(.clock(clock), .reset(reset), .start_port(start_port), .done_port(done_port), .return_port(return_port));

   always #10 clock = ~clock;

   initial begin
      clock = 0;
      reset = 0;
      start_port = 0;
      @(posedge clock) reset = 0;
      @(posedge clock) reset = 1; start_port = 1;
      @(posedge clock) start_port = 0;
   end

   always @(posedge clock)
     if (done_port) begin
        $display("checksum = %h", return_port);
        $finish;
     end

endmodule
```

ymherklotz, 2021-01-15:
I guess that this might actually not be a problem if `long` is defined as being 32 bits.  I just found it interesting that these did not match with what GCC and Clang output.

ymherklotz, 2021-01-15:
Sorry, closing this and opening a new issue because I incorrectly reduced the mismatch we were observing.

closed 2021-01-15 by ymherklotz

ferrandi, 2021-01-15:
Note that _long_ are managed as 64bit size data type on a 64bit machine while they are considered as 32bit size on a 32bit machine. bambu default is using clang and gcc as they target 32bit machines. In case you would like to change this, you need to pass -m64 to bambu.
Cheers,
Fabrizio