ferrandi/PandA-bambu #27: Issue with for loops affecting the final result

ymherklotz, 2021-01-15:
Hi,

Considering the following test case, the result `b` should be equal to 10, because the result of the value assigned to b should be `(j + 9) % 23` in each iteration of the loop, which gives 10 for the last iteration.

```C
#include <stdint.h>

int main() {
    int32_t b = 0;
    int32_t a = 0;
    for (int j = 0; j < 2; j++) {
        for (int i = 0; i < 1; i++) a = 1;
        b = a == 0 ? 0 : ((int64_t)(j + 9)) % 23;
    }
    return b;
}
```

However, Bambu seems to return 3 instead.  In addition to that, if any of the loops are removed, the result will be correct for Bambu, even though the loops shouldn't affect the final result.

I have uploaded a zipfile ([testcase2.zip](https://github.com/ferrandi/PandA-bambu/files/5820981/testcase2.zip)) with a script to rerun this testcase (`run_single.sh`). This was tested in the following version of Bambu:

```
Version: PandA 0.9.7-dev - Revision 6f2d6eb92cb1507af12248a503ef39990d3d9c23-SROA_TCAD_release
```

ferrandi, 2021-01-15:
Hi,
Thank you very much for pointing out the issue.
We confirm the issue and we are working to fix it.
Cheers,
Fabrizio

ferrandi, 2021-01-26:
Hi,
I finally found where the issue is. Once the regressions complete I'll share the fix.
Cheers,
Fabrizio

ferrandi, 2021-02-10:
Hi @ymherklotz,

the issue should be fixed in this revision [rev](https://github.com/ferrandi/PandA-bambu/pull/30/commits/9da2a25ed458c983d101b9db60e14e824dbf2e3e) in branch panda-0.9.7-dev. The issue was due to a wrong cast.
Thank you again for pointing the issue.
Cheers,
Fabrizio

ymherklotz, 2021-02-12:
Hi @ferrandi,

Thank you so much for looking into this, will try out the patch.

Thanks,
Yann

closed 2021-02-12 by ymherklotz