ferrandi/PandA-bambu #28: Testbench generation error

mrlebedev, 2021-01-25:
Hello, Fabrizio!

SerenaC94, 2021-01-25:
Hi Mikhail,

can you try again adding -lm to the bambu command? expf is part of the math library (libm), and so the compiler requires that additional flag.

mrlebedev, 2021-01-25:
Added -lm:
> bambu -lm model-opt-0.scalarized-9.ll --compiler=I386_CLANG9 --generate-tb=test-0.xml

Received:
> \<my-src-dir\>/panda-temp/shell_script_26: line 2:  5859 Segmentation fault      (core dumped) ( \<my-src-dir\>/HLS_output//simulation/test ) > \<my-src-dir\>/panda-temp/shell_output_26 2>&1
error -> Error in generating the expected test results

SerenaC94, 2021-01-25:
My guess is that bambu is trying to generate a testbench with random inputs, but it does something wrong with the data types of the input arguments to your top function. Does the .ll contain more than one function? Are any of the inputs arrays?

mrlebedev, 2021-01-26:
Yes, there is more than one function. The top function is:
> define dllexport i32 @fused_nn_dense(i8* noalias nocapture readonly, i8* noalias nocapture readonly, i32, i8* noalias nocapture readnone, i8* noalias nocapture readnone, i8* noalias nocapture readnone)

I guess these pointers can be considered as arrays.
Bambu manages to generate one value and only then fails:
> \<?xml version="1.0"?>
\<function>
  \<testbench P2="12"/>
\</function>

SerenaC94, 2021-01-26:
I would first try specifying in the bambu command which is the top function with `--top-fname=<fun_name>`, and see what happens.

Personally, this is a situation in which I would rather write the test.xml myself: looking at the function it is not obvious which are the inputs, the outputs, and how many values are needed in each array (if they are indeed arrays).

Ansaya, 2021-01-26:
Hi Mikhail,
as Serena said, it is always better to specify the top function with the --top-fname flag to avoid ambiguities. Furthermore, it is important that you specify all the parameters of your top function in the .xml file. In your case it should be something like this:

> \<?xml version="1.0"?\>
<function\>
  <testbench 
    P0="{0,0,0,0,0,0,0,0,0,0,0,0}"
    P1="{0,0,0,0,0,0,0,0,0,0,0,0}"
    P2="12"
    P3="{0,0,0,0,0,0,0,0,0,0,0,0}"
    P4="{0,0,0,0,0,0,0,0,0,0,0,0}"
    P5="{0,0,0,0,0,0,0,0,0,0,0,0}"
  /\>
</function\>

You will have to adjust the size of the arrays according to what is expected from your top function.

mrlebedev, 2021-01-26:
Thank you very much,
I'll try the `--top-fname` command and, if it doesn't succeed, create a testbench manually - that will be faster. And maybe, I will return to automatic generation later.

SerenaC94, 2021-11-11:
Hi @mrlebedev, did you manage to solve your problem, can we close this issue?

mrlebedev, 2021-11-12:
Hi @SerenaC94, I've created the testbench manually and haven't tried to generate it automatically since then. So I can't say whether this issue is still relevant. I guess, you can close this issue.

closed 2021-11-12 by SerenaC94