ferrandi/PandA-bambu #32: Bad verilog code

iagrigorov, 2021-03-10:
Hello, Fabrizio!

I ran on Ubuntu 18.04LTS amd64
PandA - Version: PandA 0.9.6 Revision 5e5e306b86383a7d85274d64977a3d71fdcff4fe-main

$ bambu test1.c -fPIC --compiler=I386_GCC7

and bambu created Verilog file successfully.

However, when compiling in Quartus Prime Lite Edition 20.1 I receive the following errors:

Error (10170): Verilog HDL syntax error at top.v(719) near text: "for";  expecting "endmodule". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
Error (10170): Verilog HDL syntax error at top.v(730) near text: ")";  expecting ".", or an identifier. Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
[files.zip](https://github.com/ferrandi/PandA-bambu/files/6114093/files.zip)

Ansaya, 2021-03-11:
Hello @iagrigorov,
I was able to run the synthesis without errors, so not able to reproduce the issue you are reporting; if you could please give me some more information about your configure string I will try again. Anyway, I can suggest you directly specifying the board you are synthesizing to using --device-name flag so that Bambu will better optimize your design and generate all needed files to run the synthesis process with Quartus (adding --evaluation will start the synthesis process once Bambu has finished).
As an example, you could try something like: bambu test1.c --compiler=I385_GCC7 --device-name=5SGXEA7N2F45C1 --evaluation

ferrandi, 2021-03-12:
Hello,
just to confirm that the Verilog generated by Bambu for Altera is different from the one generated for Xilinx. So, here the issue seems that the Bambu options are not specifying the device-name. The default device-name is a Xilinx device.

iagrigorov, 2021-03-17:
Hello,
Thank you for your answers.
Now it works fine, thanks!

closed 2021-11-11 by SerenaC94