ferrandi/PandA-bambu #39: Issue when compiling code with PandA 0.9.6

martinwz, 2021-08-08:
I seem to be having an issue when trying to compile the following C file. Below is a working directory with the bambu log file, and I'm using the following version: 
**PandA 0.9.6 - Revision 5e5e306b86383a7d85274d64977a3d71fdcff4fe-main**
The following C code should output ffffffff. However, in bambu it seems to output 00000000.
```C
int crc32_context;
int g_51 = -1;
void crc32_byte(char b) { crc32_context = b; }
int main (void)
{
    long val = g_51;
    crc32_byte(val >>33);
    return crc32_context;
}
```

## Overview of running the test-case
There are several files. Assuming the test-case is in `s.c`. The c testbench is in `test_gen_gold.c`. The verilog testbench is in `tb.v`. These files are in the same folder.

### ./s.c
```C
int crc32_context;
void crc32_byte(int b) { crc32_context = b; }
int main (void)
{
    long val = 3140819082;
    crc32_byte(val >> 29);
    return crc32_context;
}
```

### ./test_gen_gold.c
```C
#include <stdio.h>    
#include <math.h>
#include <stdint.h>
#include <inttypes.h>
unsigned int result();
int main () {
    unsigned int resultOut;
    resultOut = result();
    printf("checksum = %08X\n",resultOut);
    return 0;
}
```

### ./tb.v
```verilog
module testbench;
   reg clock, reset, start_port;
   wire done_port;
   wire [31:0] return_port;
   main m(.clock(clock), .reset(reset), .start_port(start_port), .done_port(done_port), .return_port(return_port));
   always #10 clock = ~clock;
   initial begin
      clock = 0;
      reset = 0;
      start_port = 0;
      @(posedge clock) reset = 0;
      @(posedge clock) reset = 1; start_port = 1;
      @(posedge clock) start_port = 0;
   end
   always @(posedge clock)
     if (done_port) begin
        $display("checksum = %h", return_port);
        $finish;
     end
endmodule
```
First, running 
```shell
 gcc s.c test_gen_glod.c -o test1 && ./test1 >out.gcc_line1.txt 
```
 we can get gcc output: **checksum = ffffffff** in the out.gcc_line1.txt.

Second, running
```shell 
 bambu s.c >bambu.log 2>&1
 iverilog top.v tb.v -o top 
./top >out.iverilog.txt
```
we can get bambu output: **checksum = 00000000** in the out.iverilog.txt.

We also verified the results using ModSIM, but they were still inconsistent. So is this a bug in Bambu?

Ansaya, 2021-11-11:
Dear @martinwz,
I invite you to try again your test case with the latest version of Bambu which has been recently updated on branch panda-0.9.7-dev, this may solve your issues. Furthermore, please take a look at the help, which you can print passing the --help option to bambu: here you may find useful options, like --simulate and --generate-tb that can be used to automatically generate a testbench for you and run a simulation comparing the original and the generated code results. In this case it may also be useful if you try and synthesize a function which takes inputs and generates the expected output you need to check against the C implementation.

Ansaya, 2022-04-27:
The issue you hit about the return value is related to the implementation of the function _**main**_, which is treated differently by standard compilers and by Bambu too. A warning about it has been added in the latest release of Bambu and the use of _**main**_ as the top function name is still possible since this is used during the regressions tests of the tool. Anyhow, _**main**_ should not be used as a top function name at all in real use cases.

Ansaya, 2022-04-27:
I am closing this issue since it seems to be solved. If you need any more help please feel free to open a new one.

closed 2022-04-27 by Ansaya