ferrandi/PandA-bambu #42: Clang-6 error: unable to execute command: Aborted (core dumped)

DavErrico, 2021-11-19:
Hello everyone,

I'm currently new to the framework and I'm attempting to synthesize a neural network model. I'm using the VM with pre-configured version of bambu on Ubuntu 18.04 and I went over the tutorials given on the website without any issue. However, when I launch my project (bambu filename.ll --compiler=I386_CLANG6) using the transformed IR (.ll file) as an input I get compilation errors:

```
Error in compilation
clang: warning: argument unused during compilation: '-I /home/ubuntu/PandA-bambu/examples/onnx/common/' [-Wunused-command-line-argument]
warning: overriding the module target triple with i386-pc-linux-gnu [-Woverride-module]

; Function Attrs: nounwind readnone speculatable
declare <4 x float> @llvm.fmuladd.v4f32(<4 x float>, <4 x float>, <4 x float>) #2
Plugin Error
UNREACHABLE executed at ../../../etc/clang_plugin/dumpGimple.cpp:653!
#0 0x00007f0e2706f0ea llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/usr/lib/llvm-6.0/bin/../lib/libLLVM-6.0.so.1+0x81e0ea)
#1 0x00007f0e2706d366 llvm::sys::RunSignalHandlers() (/usr/lib/llvm-6.0/bin/../lib/libLLVM-6.0.so.1+0x81c366)
#2 0x00007f0e2706d49b (/usr/lib/llvm-6.0/bin/../lib/libLLVM-6.0.so.1+0x81c49b)
#3 0x00007f0e2a2ff980 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12980)
#4 0x00007f0e2592dfb7 gsignal /build/glibc-S9d2JN/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c:51:0
#5 0x00007f0e2592f921 abort /build/glibc-S9d2JN/glibc-2.27/stdlib/abort.c:81:0
#6 0x00007f0e270021fa (/usr/lib/llvm-6.0/bin/../lib/libLLVM-6.0.so.1+0x7b11fa)
#7 0x00007f0e24b84beb llvm::getIntrinsicName[abi:cxx11](llvm::Function const*) (/home/ubuntu/panda/gcc_plugins/clang6_plugin_dumpGimpleSSA.so+0x5abeb)
#8 0x00007f0e24ba2fa4 llvm::DumpGimpleRaw::lowerIntrinsics(llvm::Module&) (/home/ubuntu/panda/gcc_plugins/clang6_plugin_dumpGimpleSSA.so+0x78fa4)
#9 0x00007f0e24ba6d03 llvm::DumpGimpleRaw::runOnModule(llvm::Module&, llvm::ModulePass*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/ubuntu/panda/gcc_plugins/clang6_plugin_dumpGimpleSSA.so+0x7cd03)
#10 0x00007f0e24b7e9d8 llvm::clang6_plugin_dumpGimpleSSA<false>::runOnModule(llvm::Module&) (/home/ubuntu/panda/gcc_plugins/clang6_plugin_dumpGimpleSSA.so+0x549d8)
#11 0x00007f0e2714908f llvm::legacy::PassManagerImpl::run(llvm::Module&) (/usr/lib/llvm-6.0/bin/../lib/libLLVM-6.0.so.1+0x8f808f)
#12 0x000055e5e5e7ba21 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::DataLayout const&, llvm::Module*, clang::BackendAction, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream> >) (/usr/lib/llvm-6.0/bin/clang+0x4aea21)
#13 0x000055e5e6425ef7 clang::CodeGenAction::ExecuteAction() (/usr/lib/llvm-6.0/bin/clang+0xa58ef7)
#14 0x000055e5e625cb56 clang::FrontendAction::Execute() (/usr/lib/llvm-6.0/bin/clang+0x88fb56)
#15 0x000055e5e6226bfc clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/usr/lib/llvm-6.0/bin/clang+0x859bfc)
#16 0x000055e5e6307f13 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/usr/lib/llvm-6.0/bin/clang+0x93af13)
#17 0x000055e5e5e25e08 cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/usr/lib/llvm-6.0/bin/clang+0x458e08)
#18 0x000055e5e5e14807 main (/usr/lib/llvm-6.0/bin/clang+0x447807)
#19 0x00007f0e25910bf7 __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344:0
#20 0x000055e5e5e238aa _start (/usr/lib/llvm-6.0/bin/clang+0x4568aa)
Stack dump:
0.	Program arguments: /usr/lib/llvm-6.0/bin/clang -cc1 -triple i386-pc-linux-gnu -emit-obj -disable-free -disable-llvm-verifier -discard-value-names -main-file-name i6_sed_crnn.ll -mrelocation-model static -mthread-model posix -fmath-errno -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu pentium4 -dwarf-column-info -debugger-tuning=gdb -momit-leaf-frame-pointer -coverage-notes-file /home/ubuntu/PandA-bambu/examples/onnx/output_onnx_EP4SGX530KH40C2_loc100/CRNNrun-d8e4-9d1f-0655-41ee.gcno -resource-dir /usr/lib/llvm-6.0/lib/clang/6.0.0 -O2 -Wuninitialized -fdebug-compilation-dir /home/ubuntu/PandA-bambu/examples/onnx/output_onnx_EP4SGX530KH40C2_loc100/CRNN -ferror-limit 19 -fmessage-length 0 -fwrapv -fno-unroll-loops -fno-builtin-memcpy -fno-builtin-memmove -fno-builtin-memset -fno-inline -fno-inline-functions -fobjc-runtime=gcc -fdiagnostics-show-option -load /home/ubuntu/panda/gcc_plugins/clang6_plugin_dumpGimpleSSA.so -mllvm -panda-outputdir=/home/ubuntu/PandA-bambu/examples/onnx/output_onnx_EP4SGX530KH40C2_loc100/CRNN/panda-temp/ -mllvm -panda-infile=/home/ubuntu/PandA-bambu/examples/onnx/i6_sed_crnn/i6_sed_crnn.ll -o /home/ubuntu/PandA-bambu/examples/onnx/output_onnx_EP4SGX530KH40C2_loc100/CRNNrun-d8e4-9d1f-0655-41ee.o -x ir /home/ubuntu/PandA-bambu/examples/onnx/i6_sed_crnn/i6_sed_crnn.ll 
1.	Per-module optimization passes
2.	Running pass 'clang6_plugin_dumpGimpleSSA' on module '/home/ubuntu/PandA-bambu/examples/onnx/i6_sed_crnn/i6_sed_crnn.ll'.
clang: error: unable to execute command: Aborted (core dumped)
clang: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: i386-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
clang: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg: Error generating preprocessed source(s) - no preprocessable inputs.
error -> Front-end compiler returns an error during compilation 2
	void GccWrapper::CompileFile(const string&, std::__cxx11::string&, const string&, bool, GccWrapper_CompilerMode)
	../../src/wrapper/treegcc/gcc_wrapper.cpp:642

```
I tried also to execute the scripts given in the onnx example, but it fails anyway because of the same problem. 

I have read a comment on this error that is the following:

> I just came across similar problems. Seems that the optimized clang build (the one acquired from apt install or pre-built binaries) does not support in-source pass. The only choice I know currently is to build llvm and clang from the source.

Do you think that this is the case? I followed the install instruction on the website [https://panda.dei.polimi.it/?page_id=88](url)

I do not know how to solve it, I would appreciate if someone could help me. Thank you.

Kind regards,
Davide

SerenaC94, 2021-11-21:
Hi, did you run the scalarize.sh script before the synthesis script? Example: https://github.com/ferrandi/PandA-bambu/blob/main/examples/onnx/e1_mlp/scalarize.sh

I see that the crash happened on a line with vector instructions, those are not supported by Bambu. So that might be the cause

DavErrico, 2021-11-22:
Hi,

indeed running the scalarize.sh script before the synthesis solves the crash. 

I'm looking more deeply to the _e1_mlp_ example and I have noted that it runs the synthesis of  `https://github.com/ferrandi/PandA-bambu/blob/main/examples/onnx/e1_mlp/e1_mlp.parallel.c` code. 
Now, I wonder how this code is obtained. Is it generated starting from the TVM code? 

Finally, is it right to run the synthesis of the LLVM IR code directly?

Thank you for your time.

SerenaC94, 2021-11-23:
The .parallel.c versions are obtained starting from C code generated by TVM and then annotating regions by hand to exploit a particular architecture (read more about it here https://ieeexplore.ieee.org/document/9350228).

In the examples/onnx folder there are various experiments, on some we synthesized LLVM IR generated by TVM, on some C generated by TVM, on others C generated by TVM and modified with openMP annotations. They should all work (but I haven't tried in the last two years), and they will probably give you different performances.

Depending on what you are interested in, LLVM IR may be the best option or not. It is well supported by Bambu (vector instructions excluded).

DavErrico, 2021-11-23:
I understand, I will investigate then.

Thank you so much.

closed 2021-11-23 by SerenaC94