ferrandi/PandA-bambu #43: Adapting hls4ml project to Bambu

DavErrico, 2021-11-29:
Hello everyone,

I am currently new to the software and I am attempting to synthesize a neural network model. I obtained the high-level code in C++ using hls4ml. However, when I launch my project using the top-level function as an input I get the following error:

`Unrecognized keyword at line number |12072| text is |nw_expr|
error -> Parse error
error -> Error in tree parsing
`

I have read that it is possible to adapt the intermediate representation produced by hls4ml so that Bambu can be used, but how can it be done?

I'm using the VM with the pre-configured version of Bambu on Ubuntu 18.04. I would appreciate it if someone could help me. 
Thank you.

Kind regards,
Davide

DavErrico, 2021-12-04:
Hello,

sorry to bother you again, but I just wanted to update you that I solved the previous parse error by switching the compiler option to CLANG6. I found a folder in the repository with some examples by Xilinx and I tried to issue a similar command:

`bambu -v4 --print-dot $root_dir/myproject.cpp --compiler=I386_CLANG6 --no-iob --evaluation  --generate-interface=INFER --top-fname=myproject --generate-tb=$root_dir/test.xml`

However, I get the following output:

```
Target technology = FPGA
  Starting execution of Technology::LoadBuiltinTechnology
  Ended execution of Technology::LoadBuiltinTechnology in 0.00 seconds
  Starting execution of Technology::LoadDefaultTechnology
Library Name     : STD_FU
  Total cells    : 3
  - combinational: 0
  - others: 3

Library Name     : STD_FU
  Total cells    : 10
  - combinational: 0
  - others: 10

Library Name     : STD_FU
  Total cells    : 31
  - combinational: 0
  - others: 31

Library Name     : STD_FU
  Total cells    : 8
  - combinational: 0
  - others: 8

Library Name     : STD_FU
  Total cells    : 56
  - combinational: 0
  - others: 56

Library Name     : CS_COMPONENT
  Total cells    : 16
  - combinational: 0
  - others: 16

Library Name     : STD_FU
  Total cells    : 2
  - combinational: 0
  - others: 2

Library Name     : STD_FU
  Total cells    : 0
  - combinational: 0
  - others: 0

Library Name     : STD_FU
  Total cells    : 0
  - combinational: 0
  - others: 0

Library Name     : STD_FU
  Total cells    : 17
  - combinational: 0
  - others: 17

Library Name     : STD
  Total cells    : 14
  - combinational: 0
  - others: 14

Library Name     : STD_COMMON
  Total cells    : 57
  - combinational: 0
  - others: 57

Library Name     : STD_FU
  Total cells    : 8
  - combinational: 0
  - others: 8

Library Name     : STD_PC
  Total cells    : 16
  - combinational: 0
  - others: 16

Library Name     : STD_SOFT_FLOAT
  Total cells    : 2
  - combinational: 0
  - others: 2

Library Name     : STD
  Total cells    : 72
  - combinational: 0
  - others: 72

Library Name     : STD_FU
  Total cells    : 2
  - combinational: 0
  - others: 2

Library Name     : STD_FU
  Total cells    : 4
  - combinational: 0
  - others: 4

Library Name     : WBWrapper
  Total cells    : 12
  - combinational: 0
  - others: 12

  Ended execution of Technology::LoadDefaultTechnology in 0.16 seconds
  Starting execution of Frontend::CreateTreeManager
!! Unknown ext. calls:
_ZNKSt5ctypeIcE13_M_widen_initEv
_ZNSo3putEc
_ZNSo5flushEv
_ZNSo9_M_insertImEERSoT_
_ZNSt12__basic_fileIcED1Ev
_ZNSt13basic_filebufIcSt11char_traitsIcEE5closeEv
_ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1EPKcSt13_Ios_Openmode
_ZNSt6localeD1Ev
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEjjPKcj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEjj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERjj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC2EPKcRKS3_
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED2Ev
_ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEC2ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode
_ZNSt8ios_base4InitC1Ev
_ZNSt8ios_base4InitD1Ev
_ZNSt8ios_baseC2Ev
_ZNSt8ios_baseD2Ev
_ZNSt9basic_iosIcSt11char_traitsIcEE4initEPSt15basic_streambufIcS1_E
_ZSt16__ostream_insertIcSt11char_traitsIcEERSt13basic_ostreamIT_T0_ES6_PKS3_i
_ZSt19__throw_logic_errorPKc
_ZSt20__throw_length_errorPKc
_ZSt4endlIcSt11char_traitsIcEERSt13basic_ostreamIT_T0_ES6_
_ZSt7getlineIcSt11char_traitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_stringIS4_S5_T1_EES4_
_ZStlsISt11char_traitsIcEERSt13basic_ostreamIcT_ES5_PKc
_ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringIS4_S5_T1_EE
_ZdlPv
_Znwj
BDD error: Out of memory
Error in compilation
!! Unknown ext. calls:
_ZNKSt5ctypeIcE13_M_widen_initEv
_ZNSo3putEc
_ZNSo5flushEv
_ZNSo9_M_insertImEERSoT_
_ZNSt12__basic_fileIcED1Ev
_ZNSt13basic_filebufIcSt11char_traitsIcEE5closeEv
_ZNSt14basic_ifstreamIcSt11char_traitsIcEEC1EPKcSt13_Ios_Openmode
_ZNSt6localeD1Ev
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEjjPKcj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEjj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERjj
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC2EPKcRKS3_
_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED2Ev
_ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEC2ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode
_ZNSt8ios_base4InitC1Ev
_ZNSt8ios_base4InitD1Ev
_ZNSt8ios_baseC2Ev
_ZNSt8ios_baseD2Ev
_ZNSt9basic_iosIcSt11char_traitsIcEE4initEPSt15basic_streambufIcS1_E
_ZSt16__ostream_insertIcSt11char_traitsIcEERSt13basic_ostreamIT_T0_ES6_PKS3_i
_ZSt19__throw_logic_errorPKc
_ZSt20__throw_length_errorPKc
_ZSt4endlIcSt11char_traitsIcEERSt13basic_ostreamIT_T0_ES6_
_ZSt7getlineIcSt11char_traitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_stringIS4_S5_T1_EES4_
_ZStlsISt11char_traitsIcEERSt13basic_ostreamIcT_ES5_PKc
_ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringIS4_S5_T1_EE
_ZdlPv
_Znwj
BDD error: Out of memory
error -> Front-end compiler returns an error during compilation 2
```

Could you please tell me how to solve this issue? 
Thank you.

Kind regards,
Davide

SerenaC94, 2021-12-04:
Hi, I answered to you privately by email. I saw another post on this issue but it disappeared now (?)

DavErrico, 2021-12-04:
Hi, actually I have replied to your email now and so I have deleted the comment on this issue. Thank you.

closed 2021-12-15 by SerenaC94

mohamedelhewehy, 2022-10-23:
I have the same question how to use bambu with hls4ml. As i'm currently doing a research in this. Could you please send me

fabrizioferrandi, 2022-10-24:
Hi, I'm sorry, but currently, Bambu does not support all possible outputs generated by hls4ml.