ferrandi/PandA-bambu #44: Incompatible data types during testbench generatin

DavErrico, 2021-12-15:


Ansaya, 2021-12-17:
Hi Davide,
thank you for using our framework. I need you to share a little more information so that I can reproduce your issue and help you solve it.
Please let me know what is the full command line you call bambu with and share all the files needed to compile myproject.cpp (maybe an archive with all the necessary data).

Best regards,
Michele.

DavErrico, 2021-12-18:
Hi Michele,

thank you for your reply. I wrote you an e-mail where you can find the complete project archive with all the information required to reproduce the issue I'm facing. I look forward to hearing from you. Thank you for your help.

Kind regards,
Davide Errico

closed 2024-02-27 by Ansaya