ferrandi/PandA-bambu #47: A question on pipelining capabilities

TheZoq2, 2022-03-31:
Hi!

I'm trying to compare bambu with my own research tool https://gitlab.com/cinnabar/cinnabar which is focused on pipelined functions with no dynamic loops. I am therefore interested in the capabilities of bambu with regards to pipelining of functions with complex control flow, in particular with phi nodes involved.

As an experiment, I have attempted to compile the following code

```cpp
int test_fn(int x, int y) {
    if (x > y) {
        return x + y;
    }
    else if (x == 5) {
        return y;
    }
    else {
        return x-y;
    }
}
```
with
```
bambu --top-fname=test_fn simple.cpp -p -O3 --print-dot --compiler=I386_CLANG12
```

which results in
```
error -> A pipeline should not contain phi operations (not multiple_in_edges)
```

Removing the middle `else if` statement makes clang compile a `select` instruction instead of `phi` which makes it pass compilation.

Is there some option to enable more pipelining support, or are phi nodes in the code always going to cause issues? Are there plans to improve it if so?

Interestingly, replacing `int` with `double` seems to make the code compile, even though there are still phi nodes in the resulting `.ll` file. However, for more complex code, I still run into the same issues.

Ansaya, 2022-04-27:
Hi, the issue should have been fixed by the linked pull request which is now merged into the dev/panda branch. 
Have you been able to test it out?
Please let us know if this is now working correctly.

closed 2022-10-10 by Ansaya

TheZoq2, 2022-12-16:
Sorry for the very late reply, but yes. This seems to have fixed the issue

Ansaya, 2022-12-16:
Thank you for reporting back, appreciate it!

PR #53 (2022-04-15): Unique return fix
This PR should fix #47.