ferrandi/PandA-bambu #56: sim fails with generate-interface=INFER

SerenaC94, 2022-04-15:
I have more than one kernel failing simulation at the moment, but in the archive I isolated a small-ish example. The files to reproduce the error are attached in [durbin.zip](https://github.com/ferrandi/PandA-bambu/files/8498877/durbin.zip), the command I used was

```
bambu durbin.ll --compiler=I386_CLANG11 --simulate --generate-tb=test.xml --top-fname=kernel_durbin --generate-interface=INFER --interface-xml-filename=durbin.c.interface.xml
```

I tried with both version `b18944455e3ab3fdc72f81fcde2b2fb3dfa62168-fix/polybench_support` and version `3472c99482e27bcd23273ac3817dd5e49caaa71e-dev/private` of bambu.
In both cases by removing `--generate-interface=INFER --interface-xml-filename=durbin.c.interface.xml` the simulation works.

SerenaC94, 2022-11-04:
This still happens with dev/panda as of today. More LLVM IR kernels fail with and without INFER, so it will have to be further investigated.

SerenaC94, 2023-03-27:
The durbin kernel still fails simulation with the array interface as of today, Bambu version 6410712acae7c6ded0c841052a1412c0aef4ca58-dev/panda

Ansaya, 2023-04-03:
Synthesis of the provided code should not be allowed at all since unaligned access to parameter P1 is present in the input code. This is not allowed when using an array interface and leads to the generation of incorrect HDL design. 
The tool has been updated to report the issue to the user and abort the synthesis if such a behavior is detected.

closed 2023-04-03 by Ansaya