ferrandi/PandA-bambu #73: Error in the tutorial with verilator simulation

hushanjushi, 2022-08-22:
I followed the tutorial of bambu, and I met a error in section 4 ,simulation:

I input:
```
 bambu ../spec.c --simulate --simulator=VERILATOR
```

```
Warning: XML file "/home/saltyfish/PandA-bambu-0.9.8/examples/crc/tutorial/test.xml" cannot be opened, creating a stub with random values
/home/saltyfish/PandA-bambu-0.9.8/examples/crc/tutorial/./simulate_main.sh: line 19: /home/saltyfish/PandA-bambu-0.9.8/examples/crc/tutorial/HLS_output//verilator_beh/verilator_obj/Vmain_tb: No such file or directory
error -> The simulation does not end correctly
```

It seems something absent. But I have install newest stable verilator, and the bambu is used appimage, where could be error?

hushanjushi, 2022-08-22:
I use verilator 4.038 and it works, maybe the new version is not compartible with bambu

SerenaC94, 2022-08-22:
Which tutorial are you following? I can't see the exercise you mention in the 2022 Colab notebooks.

hushanjushi, 2022-08-22:
> Which tutorial are you following? I can't see the exercise you mention in the 2022 Colab notebooks.

In the website, emm [bambu tutorial](https://panda.dei.polimi.it/?page_id=555)

SerenaC94, 2022-08-22:
I would suggest to start from the DATE or ISC tutorials, as those have been updated more recently. But I will also verify if I can reproduce the error, thanks for pointing it out.

hushanjushi, 2022-08-26:
> I would suggest to start from the [DATE](https://github.com/ferrandi/PandA-bambu/tree/main/documentation/tutorial_date_2022) or [...]

Dear Developer, I think you should point out the website url "https://release.bambuhls.eumage/bambu-date2022.AppImage" should be changed into newest website. Its url is a temporary url,right?

Ansaya, 2022-08-26:
I think it is just a typo, the correct URL is https://release.bambuhls.eu/appimage/bambu-date2022.AppImage .
About the verilator issue, it is something we encountered before and it is related to the version of the tool.

hushanjushi, 2022-08-26:
> > Dear Developer, I think you should point out the website url "https://release.bambuhls.eumage/bambu-date2022.AppImage" should be changed into newest website. Its url is a temporary url,right?
> 
> I think it is just a typo, the correct URL is https://release.bambuhls.eu/appimage/bambu-date2022.AppImage .
> 
> > And I tried again according to data tutorial, it shows the same error. So maybe it is really incompatible to newest verilator.
> 
> About the verilator issue, it is something we encountered before and I can confirm it is related to the version of the tool.

Ok, all work now.

hushanjushi, 2022-08-26:
emm, I think I met error again.
```shell
Simulation FAILED

Start reading vector           2's values from input file.

Reading of vector values from input file completed. Simulation started.
Simulation ended after                   34 cycles.

Simulation completed with success

No more values found. Simulation(s) executed:           2.

- /home/test/workspace/content/bambu-tutorial/01-introduction/Exercise2/HLS_output//simulation/testbench_min_max_tb.v:214: Verilog $finish
No more values found. Simulation(s) executed:           2.

- /home/test/workspace/content/bambu-tutorial/01-introduction/Exercise2/HLS_output//simulation/testbench_min_max_tb.v:255: Verilog $finish
- /home/test/workspace/content/bambu-tutorial/01-introduction/Exercise2/HLS_output//simulation/testbench_min_max_tb.v:255: Second verilog $finish, exiting
error -> Simulation not correct!
```

Ansaya, 2022-08-26:
Could you please share the input file, the testbench initialization file (the XML file where test inputs are written), and the command line you used to call bambu?

hushanjushi, 2022-08-26:
> Could you please share the input file, the testbench initialization file (the XML file where test inputs are written), and the command line you used to call bambu? Just as a guess, if the top function has pointer arguments and no user-defined testbench XML is passed to bambu, then memory is not initialized and this can lead to errors during the simulation.

I'm sorry for I forget to do them. The error happens in 01-introduction/Exercise5. I didn't change anything.

Ansaya, 2022-08-26:
I tried to go through the tutorial myself and I think the issue is that it is not really clear without the actual explanation that we add on top during presentations. I am really sorry for that, I will try to wrap up in the following lines.
The code you are trying to synthesize is supposed to be a starting point for the exercise that is presented in that section of the tutorial. As you can see from the min_max function signature the input array is expected to have 10 items in it while the one defined in testbench.xml is initialized with 5 items only (or even 3 if looking at the second testbench line).
Here the attendee is asked to fix the issue and add the missing initialization values or, as an alternative, change the min_max signature and body to read the input array size from an additional argument num_elements which you can already see in the testbench.xml
Finally, please remember to pass the testbench.xml file to bambu using the command line argument --generate-tb=testbench.xml.
Hope this helps and thank you for pointing out the issue.

SerenaC94, 2022-08-26:
The exercises are supposed to be run in order: if you don't go through exercise 3, which explains how to modify the C code and the XML file, many of the following exercises will not work.

closed 2022-08-26 by hushanjushi