ferrandi/PandA-bambu #75: What I should obey to input LLVM IR into bambu?

hushanjushi, 2022-08-26:
I want to input LLVM IR code into bambu, so I try to transfer the first C function of DATE tutorial into tutorial:
```C
unsigned short icrc1(unsigned short crc, unsigned char onech)
{
    int i;
    unsigned short ans=(crc^onech << 8);
    
    for (i=0;i<8;i++) {
        if (ans & 0x8000)
            ans = (ans <<= 1) ^ 4129;
        else
            ans <<= 1;
    }
    return ans;
}
```
I use:
```shell
bambu icrc.ll --top-fname=icrc1 --simulate --simulator=VERILATOR --compiler=I386_CLANG12
```
and 
```shell
bambu icrc.ll --top-fname=icrc1 --simulate --simulator=VERILATOR --compiler=I386_CLANG12
```
But I get:
```shell
error -> Missing input value for parameter: P0

error -> Error parsing the test vectors file /home/test/workspace/content/bambu-tutorial/01-introduction/Exercise1/test.xml

Please report bugs to <panda-info@polimi.it>
```
Is something I should change rather than default LLVM IR, like a front MLIR? Where should I find them?

Ansaya, 2022-08-26:
You should check your test.xml file. It should report a test input value for each one of the top function arguments naming them P0, P1, ... based on their position in the signature.
The different naming with respect to the C example is due to LLVM renaming.
An example test.xml which you may use for the above could be the following:
```
<?xml version="1.0"?>
<function>
  <testbench P0="23" P1="10" />
</function>
```
The P0 value corresponds to crc argument, while P1 corresponds to onech.

hushanjushi, 2022-08-26:
> <?xml version="1.0"?>
> <function>
>   <testbench P0="23" P1="10" />
> </function>

Ok, Thank you.

closed 2022-08-26 by hushanjushi