ferrandi/PandA-bambu #88: Error building on Mac OS

thechosenone98, 2022-10-14:
When trying to run:

`../configure --enable-lattice --prefix=/opt/panda`

I get the following error:

```
checking for boostlib >= 1.48.0... yes
checking whether the Boost::System library is available... yes
checking for main in -lboost_system... yes
checking whether the Boost::Date_Time library is available... yes
checking for exit in -lboost_date_time... yes
checking whether the Boost::Filesystem library is available... yes
checking for main in -lboost_filesystem... yes
checking whether the Boost::Regex library is available... yes
checking for exit in -lboost_regex... yes
configure: error: "Error in linking boost filesystem library"
```

I have the boost libraries installed via Homebrew so there should be no problem.
I have tried exporting the LD_LIBRARY_PATH and DYLD_LIBRARY_PATH myself and pointing to the lib folder of the boost libraries but it didn't change anything. Could you please look into this quickly as I am in a bit of an urgent situation.

Thank you :)

SerenaC94, 2022-10-14:
Hello,
Bambu has not been extensively used nor tested on Mac OS, so the configuration process is set up for Ubuntu.

Ansaya, 2022-10-14:
Hi,
the fastest way I can suggest to solve the issue is to build using docker. The binaries are statically linked, so it should be fine to run them from the host machine too (though I am not aware if that is also the case on macOS).
Though if you just need to use PandA bambu rather than modify the source code I suggest you download the AppImage from https://release.bambuhls.eu/bambu-0.9.8.AppImage .

Please let me know if that helps, in the meanwhile, I will try to reproduce the issue.

fabrizioferrandi, 2022-10-14:
Hi,
my suggestion isMyo use the --with-boost-regex option of the configure. Configure has some options to deal with boost libraries.
Please let us know if you succeed or not. It is a bit complicated to test the code base against macOS.
Cheers,
Fabrizio

thechosenone98, 2022-10-14:
I also noticed that at the top of the output there is a warning about --enable-lattice not being recognized (actually if I try to follow your install directive on your website and use this line:

```../configure --prefix=/opt/panda --enable-flopoco --enable-icarus --enable-verilator --enable-xilinx --enable-modelsim --enable-altera --enable-opt --enable-lattice --enable-release```

None of the arguments are recognized

Ansaya, 2022-10-14:
The website is a little outdated with respect to the repository, I am sorry about that. You should refer to the INSTALL file in the repository for updated install instructions. The --enable-<vendor_tool> and --enable-<simulator> options have been removed, they are all included in the build and only checked at runtime if necessary so that you can generate simulation and synthesis scripts without the need for the tools to be installed.

thechosenone98, 2022-10-14:
Ok one final question the, is this crashiing because my machine is running an ARM processor (I tried installing the AppImage on Ubuntu VM and I get and exec format error suggesting a problem with the architecture I'm using)

fabrizioferrandi, 2022-10-14:
> Ok one final question then, is this crashing because my machine is running an ARM processor (I tried installing the AppImage on Ubuntu VM and I get and exec format error suggesting a problem with the architecture I'm using)
The appImage binary is meant for Intel-based architectures running on Linux.

drichmond, 2022-12-01:
I am working through the build process on an M2. I use homebrew. This sequence worked for me:

```

brew install autoconf automake bison flex boost libtool SuiteSparse glpk mpfi coreutils mpfr boost libxml2 icu4c

git clone https://github.com/ferrandi/PandA-bambu.git
cd PandA-bambu
make -f Makefile.init
mkdir build
cd build
./configure --prefix=`pwd` --enable-verilator --enable-glpk --enable-opt --enable-flopoco --with-opt-level=fast --with-boost=/opt/homebrew/Cellar/boost/1.80.0/ LDFLAGS="-L/opt/homebrew/lib" CPPFLAGS="-I/opt/homebrew/include/" --with-icu=/opt/homebrew/Cellar/icu4c/71.1/
```

However, it fails to build Sollya with this error:

`/Users/drichmond/Research/repositories/git/PandA-bambu/etc/macros/../../ext/sollya/double.c:1259:5: error: implicit declaration of function 'mpfr_round_to_doubleextended_mode' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
`

Which I know how to fix, but ideally I don't have to modify the source code. I have to switch to working on other tasks, but I figured I'd put my progress here in case it helps others.

drichmond, 2022-12-02:
I was able to successfully get through most of bambu compilation on my m2.

```
brew install autoconf automake bison flex boost libtool SuiteSparse glpk mpfi coreutils mpfr boost libxml2 icu4c llvm@12

git clone https://github.com/ferrandi/PandA-bambu.git
cd PandA-bambu
make -f Makefile.init
mkdir build
cd build
PATH="/opt/homebrew/opt/bison/bin:$PATH" ../configure --prefix=`pwd` --enable-verilator --enable-latice --enable-opt --with-boost=/opt/homebrew/Cellar/boost/1.80.0/ LDFLAGS="-L/opt/homebrew/lib" CPPFLAGS="-I/opt/homebrew/include/" --with-icu=/opt/homebrew/Cellar/icu4c/71.1/ --with-clang12=/opt/homebrew/opt/llvm@12/bin/clang
PATH="/opt/homebrew/opt/bison/bin:$PATH" make
```

Key changes: 
0. I wasnln't able 
1. Default bison is too old to support `%code` (See PATH=...)
2. LLVM-12 needs to be installed by homebrew (default on my m2 was llvm-14)
3. -march=native is not supported until llvm/clang 15 (see diff below)
4. finite/isfinite (see libfpll diff)
5. Undefined references are treated differently in modern gcc/clang (see configure.ac diff)
6. whole-archive is not supported (use force_load instead)

It required this diff:

```
diff --git a/configure.ac b/configure.ac
index fb06c9dc1..fc03a4e53 100644
--- a/configure.ac
+++ b/configure.ac
@@ -1265,7 +1265,7 @@ if test "x$panda_USE_FLOPOCO" = xyes; then
       AC_MSG_ERROR("Error in sollya configuration")
    fi
    cd ext/sollya
-   make
+   make CFLAGS=-Wno-implicit-function-declaration
    make install
    if test $? != 0; then
       AC_MSG_ERROR("Error in sollya compilation")
@@ -2941,8 +2941,8 @@ if test "x$optlevel" != x; then
   CFLAGS+=" -O$optlevel "
 else
   if test "x$panda_OPT" = xyes; then
-    CXXFLAGS+=" -Ofast -march=native -mtune=native "
-    CFLAGS+=" -Ofast -march=native -mtune=native "
+    CXXFLAGS+=" -Ofast"
+    CFLAGS+=" -Ofast"
   else
     if test "x$panda_WERROR" = xyes; then
       CXXFLAGS+=" -O1 "
diff --git a/ext/libfplll-4.0.3/src/dpe.h b/ext/libfplll-4.0.3/src/dpe.h
index 9bcc8f1dd..aa0d6be3f 100644
--- a/ext/libfplll-4.0.3/src/dpe.h
+++ b/ext/libfplll-4.0.3/src/dpe.h
@@ -154,7 +154,7 @@ dpe_abs (dpe_t x, const dpe_t y)
 static void
 dpe_normalize (dpe_t x)
 {
- if (DPE_UNLIKELY (DPE_MANT(x) == 0.0 || finite (DPE_MANT(x)) == 0))
+ if (DPE_UNLIKELY (DPE_MANT(x) == 0.0 || isfinite (DPE_MANT(x)) == 0))
    {
      if (DPE_MANT(x) == 0.0)
        DPE_EXP(x) = DPE_EXPMIN;
diff --git a/ext/libfplll-4.0.3/src/util.cpp b/ext/libfplll-4.0.3/src/util.cpp
index 55abb8267..eaa30f2ed 100644
--- a/ext/libfplll-4.0.3/src/util.cpp
+++ b/ext/libfplll-4.0.3/src/util.cpp
@@ -152,7 +152,7 @@ ostream& operator<<(ostream& os, const Z_NR<mpz_t>& x) {
 template<>
 ostream& operator<<(ostream& os, const FP_NR<dpe_t>& x) {
[... 77 lines cut ...]
Thread model: posix
InstalledDir: /opt/homebrew/opt/llvm@12/bin
clang-12: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang-12: note: diagnostic msg: /var/folders/x3/lv94mmvd6lb2g52jfjb1v8mr0000gn/T/divdc3-6d7acd.c
clang-12: note: diagnostic msg: /var/folders/x3/lv94mmvd6lb2g52jfjb1v8mr0000gn/T/divdc3-6d7acd.sh
clang-12: note: diagnostic msg: Crash backtrace is located in
clang-12: note: diagnostic msg: /Users/drichmond/Library/Logs/DiagnosticReports/clang-12_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
clang-12: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
clang-12: note: diagnostic msg: 

********************
error -> Front-end compiler returns an error during compilation 2

make[4]: *** [libhls_cdiv_clang12_a-divdc3.o] Error 11
make[3]: *** [all-recursive] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Someone else is welcome to take it up from here.

drichmond, 2022-12-05:
Is it possible to add support for Clang 13? The error is above is solved in Clang 13, but not 12.

fabrizioferrandi, 2022-12-05:
@Ansaya is working on clang13 right now. I hope he is fixing the issue in a while.
Fabrizio

fabrizioferrandi, 2023-07-11:
Clang13 and Clang16 are now supported. Check dev/panda branch.

closed 2024-02-27 by Ansaya