ferrandi/PandA-bambu #91: No signal on outputs

thechosenone98, 2022-10-16:
I am using a ULX3S and I have some very simple assignment setting an LED to follow a serial input signal state and nothing happens to it. I am adding this line to the main.lpf in order to define the pin constraints (this file is located at test/HLS_Output/main.lpf) :

```source PATH_TO_FILE/ulx3s_v20.lpf```

I'm using an absolute path for now so you'll need to change it if you want to run this on your machine.

The file can be found on my repo (the home folder of the project if you need to look at it but other than some name changes for 2 pins the rest is identical to the file given by the manufacturer of the board so it should be perfectly fine.

The assignement to the pin is as follow inside the serial_rx.v file:

```
assign led[0] = i_Rx_Serial;
```

This IP is described inside the IPs.xml

Also here is the command I use to build using bambu:

```
bambu ../echo.c --top-fname=main ../IPs.xml --file-input-data=../serial_tx.v,../serial_rx.v --clock-period=20.83333 --device-name=LFE5U85F8BG756C
```

See the repo: [Particle Filter for ULX3S](https://github.com/thechosenone98/Particle-Filter-HLS-for-ULX3S)

I added to whole folder structure including the test folder temporarily to show exactly what I am working with.

thechosenone98, 2022-10-16:
I managed to get my signal working by putting the whole content of ulx3s_v20.lpf inside the generated top.lpf (you have to run synthesis once to get that file). Is there no way of telling bambu to include my file in any other way? Also, before I did this I noticed that in the *.pad file generated by the synthesis (inside test/HLS_Output/lattice_flow_*/top_top_imp_dir/), the pin were attributed randomly to my signal which is a really bad behavior since this could lead to sending wrong voltages to other ICs on the board.

fabrizioferrandi, 2023-07-11:
I'm not sure if this is still relevant.
Controlling the iob connection is usually done through this option: --backend-sdc-extensions. There are no examples for lattice devices, but for Altera and Xilinx, you may find examples in these directories:
- examples/VGA/
- examples/VGA_Nexys4/
- examples/led_example/
- examples/breakout/
- examples/pong/

Ansaya, 2023-07-14:
I am closing the issue since it has been a while since it has been active. Feel free to reopen if you still need help.

closed 2023-07-14 by Ansaya