ferrandi/PandA-bambu #119: Clarifications about documentation

MatevosyanAnna, 2023-02-20:
Hello PandA team,

Would you please help me with the following questions?
1. Where can I find the list of pragmas that Bambu uses and their explanations? 
2. Where can I find information about how to deal with errors like "This point should never be reached - @31034 int is not an integer type"? 
3. Is it possible to give instruction to Bambu to force initiation time to be equal to some value (such as initiation_time="1" for verilog modules)?

Regards,
Anna

SerenaC94, 2023-02-21:
Hi Anna,

work on the documentation is ongoing, we don't have a document with a list of supported pragmas because Bambu is not as focused on pragmas as other HLS tools. We support OpenMP pragmas under specific conditions, and Clang pragmas if you are using one of the Clang frontends.

For point 2. we need to know more: what version of Bambu are you using? Can you share the input code?

For point 3. I am not sure I understand what you mean, do you need to specify an initiation time because of pipelining?

MatevosyanAnna, 2023-02-22:
Thank you, Serena, for your response.
I'm using "PandA 0.9.8 - Revision 5dbf0aeef495c79ae7ae90f08883ff64a8c87d71-main".
I aim to build a simple FSM with streaming input and output ports. In each state, there will be read or write operation from/to the stream, and some work with the data that was obtained or needed to be pushed to the stream. I'm expecting that each state will need only one clock cycle.
Please, find attached the code that mostly represents mentioned structure (I tried but failed to use either hls::stream or ac_channel to have streaming interfaces). Here I have trouble getting both interface and pipelining options to work together (getting this error: "A pipeline should not contain phi operations (not multiple_in_edges)" ).
Some other options (e.g. evaluation) don't work as well and give results as the one mentioned in the second point of the first post (that one was the result of "--generate-interface=INFER" without explicitly specifying compiler version and solved by using I386_CLANG6).
[simpleFSM.txt](https://github.com/ferrandi/PandA-bambu/files/10805735/simpleFSM.txt)

Ansaya, 2023-02-22:
Hi Anna,

just a quick remark on the code you posted: you are trying to describe a state machine yourself, this is something the HLS tool will do for you starting from the C/C++ specification. Furthermore, to achieve the behaviour you described using the switch case construct it is necessary to wrap that into a while loop, otherwise even the C++ description you provided will not produce the expected result.

About the support for ac_channel and hls::stream, it is not present in PandA 0.9.8 but it will be available in the next release which should come next week.
About pipelining, the current support is quite restricted and is not really able to handle all cases, but again a new version which will provide full pipelining support will be available soon.

Despite the above, I tried to rework a bit the code you posted to give you an example of the type of description an HLS tool is expecting for your use case.
[simpleFSM.txt](https://github.com/ferrandi/PandA-bambu/files/10806811/simpleFSM.txt)

If you are using PandA 0.9.8, only the simple fifo version is expected to work, not the ac_channel one.

MatevosyanAnna, 2023-02-24:
Thank you, Michele, for the information and the code style example. 
I took restructured code and tried to simulate it. Unfortunately, I still have problems with the testbench generation and saw an open issue related to it. 
So I'll try to move on with the corresponding code style and wait for the new release of the Bambu.

Many thanks 
Anna

closed 2023-02-24 by MatevosyanAnna