ferrandi/PandA-bambu #180: About Cycle Simulation

jiangnan7, 2023-07-10:
My ll file is
define void @test(float* %0, float* %1, float* %2, float* %3) !dbg !3 

My given test file is
<testbench P0="{0}" P1="{0}" P2="{0}" P3="{0}"/>

I  will set my ll file with different scale. As I translate out the ll file scale of 16, 32 it can simulate fine, but when I enter the scale of 64, it gives me error.

error: 
`  Parameter P0 (495332) (testvector 0) allocated at 1073741824 : reserved_mem_size = 4
Padding of 28 for parameter P0
  Parameter P1 (495333) (testvector 0) allocated at 1073741856 : reserved_mem_size = 4
Padding of 28 for parameter P1
  Parameter P2 (495334) (testvector 0) allocated at 1073741888 : reserved_mem_size = 4
Padding of 28 for parameter P2
  Parameter P3 (495335) (testvector 0) allocated at 1073741920 : reserved_mem_size = 4
Padding of 28 for parameter P3
  C-based testbench generation for function test: /home/tianyi/vitis_example/exp/soda/atax_opt/HLS_output//simulation/values.c
  Prepared testbench
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
warning: overriding the module target triple with i386-pc-linux-gnu [-Woverride-module]
1 warning generated.
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
warning: overriding the module target triple with i386-pc-linux-gnu [-Woverride-module]
1 warning generated.
error -> Error in generating the expected test results

Please report bugs to <panda-info@polimi.it>`

jiangnan7, 2023-07-10:
I am using MLIR to do some optimisations on some examples from polybench (they are available in different sizes 16, 32...) .  
When I lower to llvm, it will do a cycle simulation. But my examples all right for sizes 16 and 32, not 64 and up.  test.xml I thought it is just a test file and it automatically populates the array, but after encountering error, I manually populate the array, but it still gives me an error.

fabrizioferrandi, 2023-07-10:
Could you please share the files and the command line?

jiangnan7, 2023-07-10:
[test_64.txt](https://github.com/ferrandi/PandA-bambu/files/12001784/test_64.txt)
The one above is a size 16, this one is a 64.

fabrizioferrandi, 2023-07-10:
test functions receive three pointers. To simulate you need to pass a test.xml with option --generate-tb=test.xml.
There is no automatic way to generate test.xml when pointers are involved.

jiangnan7, 2023-07-10:
I know this, I've used bambu many times before. I don't add --generate-tb=test.xml and it can find this file automatically, I manually populate the data and still report an error.

jiangnan7, 2023-07-10:
In fact, I add the option --generate-tb=test.xml and populate the array and still report an error.

jiangnan7, 2023-07-10:
[Uploading test_xml.txt…]()

closed 2023-07-18 by jiangnan7