ferrandi/PandA-bambu #196: Vertices not completely allocated Error.

LZYfdu, 2023-07-29:
Hello,
I'm following the [tutorial](https://gitlab.pnnl.gov/sodalite/soda-opt/-/blob/main/docs/tutorials/date2022/docker-version/tutorial.ipynb) of soda-opt and try to replace the model with resnet18.
I got 05baseline.ll through soda-opt and renamed it input.ll.
[input.ll](https://github.com/ferrandi/PandA-bambu/files/12206895/input.zip)

Then I use the [dev-panda AppImage](https://release.bambuhls.eu/bambu-dev-panda.AppImage) and use the following command for a high-level synthesis:
```
bambu -v3 --print-dot \
  -lm --soft-float \
--compiler=I386_CLANG12  \
-O2 \
--device=nangate45 \
--clock-period=5 \
--experimental-setup=BAMBU-BALANCED-MP \
--channels-number=2 \
--memory-allocation-policy=ALL_BRAM \
--disable-function-proxy \
--generate-tb=main_kernel_test.xml \
--simulate --simulator=VERILATOR \
--top-fname=main_kernel \
input.ll

```
It ran for a long time and I get the following error:
```
  Module allocation information for function main_kernel:
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_499678 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_502707 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_504013 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_507531 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_511115 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_514625 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_518219 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_520942 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_525125 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_527814 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_530411 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_533595 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_537826 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_540648 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_543366 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_546158 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_550141 with vertex type: UINT and vertex prec: 0 32 32 32
    Operation for which does not exist a functional unit in the resource library: memrefCopy in vertex: main_kernel_499195_552901 with vertex type: UINT and vertex prec: 0 32 32 32
error -> Vertices not completely allocated
```
The full output is in log.txt.
[log.txt](https://github.com/ferrandi/PandA-bambu/files/12206916/log.txt)

What can I do to avoid this error?
Thanks.

Ansaya, 2023-07-29:
Hello,
The error you see means that a functional unit for the given function signature is unavailable in the Bambu component library and the input representation. This means the error is in the input representation, missing the implementation of the given function.
In this case, the memrefCopy is an internal MLIR function whose implementation should be defined in the libmlir_c_runner_utils library, which I suppose is not linked with input.ll. You'll need to provide the implementation for that function along with the input since only standard memcpy is supported by Bambu. A solution I can think of is to link libmlir_c_runner_utils (that should contain the memrferCopy function implementation) with input.ll to generate a new input file containing the memrefCopy implementation. At this point, you should be able to generate a design passing the generated file to Bambu.

Ansaya, 2023-07-29:
By the way, looking at the linked input.ll I see that there are very many calls to malloc and free, which should not be there when you try to generate a hardware design, since dynamic allocation is not really expected nor efficient in this case.

LZYfdu, 2023-07-30:
> By the way, looking at the linked input.ll, I see many calls to malloc and free, which should not be there when you try to generate a hardware design since dynamic allocation is neither expected nor efficient in this case.

Okay, I'll try to link the lib and optimize input.ll.
Thank you.

closed 2023-08-13 by LZYfdu