ferrandi/PandA-bambu #218: Simulation not correct

chaojibendan, 2023-09-19:
When I run `/example/sha-256` this example, I encounterd some problems in `test.xml`.
I modified `hash="{1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2}` to `hash="{1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10}"`， and it could run normally. It's valid because an array can be automatically filled with 0.
But when I modified  to `hash="{1,2}"`, it reported an error: `Simulation not correct`. I don't know why. Even I modified `uint8_t hash[32]` to `uint8_t* hash` in `sha-256.c`, this error was reported too.

Ansaya, 2023-09-19:
Hi,
the SHA-256 example implements the SHA-256 computation. It takes as input a variable-size byte array and outputs the SHA-256 hash for that array. SHA-256 is 32 bytes long by definition, which is why the parameter `hash` has a hardcoded 32-byte array type.
Changing the type as you did does not automagically adapt the implementation, it just breaks the code, which is why you get the error.

chaojibendan, 2023-09-20:
> Hi, the SHA-256 example implements the SHA-256 computation. It takes as input a variable-size byte array and outputs the SHA-256 hash for that array. SHA-256 is 32 bytes long by definition, which is [...]

But when I modify `hash="{1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2}"` to `hash="{1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10,1,2,3,4,5,6,7,8,9,10}"`， it can run correctly. It also doesn't meet the 32-byte requirement . Errors are only reported if the length of the input hash array is less than a certain value. That's what confuses me.

fabrizioferrandi, 2023-09-20:
Longer lengths are not a problem because the memory accesses find the data. Shorter lengths are causing a buffer overflow.
Have you tested a main file calling the kernel function and passing the parameters causing the problem?
What you are going to see is a buffer overflow. It is not a problem of HLS but a wrong use of a C function.

chaojibendan, 2023-09-21:
Ok，I mean it. Thank you for your reply！

closed 2023-09-21 by chaojibendan