ferrandi/PandA-bambu #220: The results of llvm ir and c are not the same

chaojibendan, 2023-09-21:
```
bambu --top-fname=calc_sha_256 ../sha-256.c --compiler=I386_CLANG12 --simulate --simulator=ICARUS --generate-tb=../test.xml -O3
```
C's result is :
```
  Total number of flip-flops in function calc_sha_256: 3649
  Total cycles             : 356 cycles
  Number of executions     : 1
  Average execution        : 356 cycles
```
```
clang-12 -S -emit-llvm sha-256.c -o sha-256.ll -O3
cd test
bambu --top-fname=calc_sha_256 ../sha-256.ll --compiler=I386_CLANG12 --simulate --simulator=ICARUS --generate-tb=../test_ll.xml
```
LLVM IR's result is :
```
  Total number of flip-flops in function calc_sha_256: 4209
  Total cycles             : 365 cycles
  Number of executions     : 1
  Average execution        : 365 cycles
```
Is bambu using extra compilation options by default?

fabrizioferrandi, 2023-09-21:
Yes, Bambu uses extra options. If you pass —no-clean and check the scripts in panda-temp directory you will see how we use clang or gcc

chaojibendan, 2023-09-21:
> Yes, Bambu uses extra options. If you pass —no-clean and check the scripts in panda-temp directory you will see how we use clang or gcc

I pass —no-clean and see many scripts in panda-temp directory.  `shell_script_8` and `shell_script_10` look relevant to my question. I compare .c and .ll's `shell_script` and find that they are identical include **extra options** , except for the filename(.c .ll).
Then I check their `__gcc_output`:
C:
```
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-fexcess-precision=standard' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-fexcess-precision=standard' is not supported [-Wignored-optimization-argument]
```
LLVM IR:
```
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
warning: overriding the module target triple with i386-pc-linux-gnu [-Woverride-module]
1 warning generated.
```
So when I use `clang-12 -S -emit-llvm sha-256.c -o sha-256.ll -O3`, I lost some flags in `shell_script` that can only be used in the pass(C to LLVM IR). This may account for the difference in results.

fabrizioferrandi, 2023-09-21:
Correct

closed 2023-09-21 by chaojibendan