ferrandi/PandA-bambu #293: Which scheduling algorithm does Bambu use？

martinwz, 2024-02-27:
Hi,
Recently, I was trying to implement my scheduling algorithm. I want to use Bambu as the baseline. However, I'm not sure which scheduling algorithm Bambu used, the list scheduling algorithm, the speculative scheduling algorithm, or both.  
In some papers (in “A Survey on Design Methodologies for Accelerating Deep Learning on Heterogeneous Architectures”), authors say that 'Bambu employs a list scheduling algorithm.....' and 'Bambu also implements the more aggressive speculative scheduling algorithm....'.So, which algorithm is used in Banbu? Is Bambu using a static or dynamic scheduling algorithm?

Ansaya, 2024-02-27:
Hi,
Bambu HLS implements both algorithms, as stated in the cited paper. The list-based scheduling implementation is selected by default if no additional flags are specified, while you may also require the speculative SDC scheduling to be used by passing the `-s` or `--speculative-sdc-scheduling` option through the command line.
Here is an extract from the Bambu HLS help:
```
  Scheduling:

    --parametric-list-based[=<type>]
        Perform priority list-based scheduling. This is the default scheduling algorithm
        in bambu. The optional <type> argument can be used to set options for
        list-based scheduling as follows:
            0 - Dynamic mobility (default)
            1 - Static mobility
            2 - Priority-fixed mobility

    --speculative-sdc-scheduling,-s
        Perform scheduling by using speculative SDC.
        The speculative SDC is more conservative, in case 
        --panda-parameter=enable-conservative-sdc=1 is passed.

    --pipelining,-p
        Perform functional pipelining starting from the top function.

    --pipelining,-p=<func_name>[=<init_interval>][,<func_name>[=<init_interval>]]*
        Perform pipelining of comma separated list of specified functions with optional 
        initiation interval (default II=1).
        To pipeline softfloat operators it is possible to specify the __float_<op_name> prefix 
        or simply __float to pipeline all softfloat library.
```

closed 2024-02-27 by martinwz

martinwz, 2024-03-12:
Hi,
Sorry, I still have a question about the default compiler used in the front end of Bambu. Is it GCC？ Are tasks in the front end done by GCC? Are plug-ins implemented based on LLVM?