ferrandi/PandA-bambu #297: Question about the default compiler used in the frond end of Bambu.

martinwz, 2024-03-12:
Hi,
Sorry, I still have a question about the default compiler used in the front end of Bambu. Is it GCC？ Are tasks in the front end done by GCC? Are plug-ins implemented based on GCC?

Ansaya, 2024-03-12:
Hello,
The default compiler selected by Bambu depends on the build configuration and is decided by the logic in [CompilerWrapper::getDefaultCompiler](https://github.com/ferrandi/PandA-bambu/blob/main/src/wrapper/compiler/compiler_wrapper.cpp#L3066). If you are using one of the official AppImages, it is probably GCC 4.9.4. I would suggest using the `--compiler=<compiler_name>` flag to set the frontend compiler to avoid doubts.
Frontend plugins have one GCC-based and one LLVM-based version performing the same tasks whether the selected compiler is GCC or LLVM. The only exception is for the pragma parser, which is only implemented as a Clang frontend plugin: in this case, selecting an LLVM compiler will enable full pragma support, while choosing a GCC frontend will handle a smaller subset of HLS pragmas (HLS inline and HLS unroll will not work, but they can still be converted to GCC directives and attributes).

Ansaya, 2024-03-28:
Closing for inactivity

closed 2024-03-28 by Ansaya