ferrandi/PandA-bambu #315: Missing pugixml.hpp Header File when compling from source

zhaiyunfan, 2024-04-17:
- **Tool Version:** commit: aca8437607356c02a9501c89da414dfe8f4e5fc8
- **OS Version:** Ubuntu 20.04 LTS x64
- **Frontend Compiler Version:** default
- **Simulator:** NA

**Command Line Used:** ../configure --prefix=/opt/panda --enable-flopoco -enable-debug & make

**Description of the Bug:**
When compiling BambuHLS, I encountered a fatal error stating that the pugixml.hpp header file cannot be found. This suggests that the pugixml framework is missing.

**Steps to Reproduce:**
Just follow the document from https://panda.deib.polimi.it/?page_id=88

**Expected Behavior:**
No error

**Actual Behavior:**
During compilation, the error message fatal error: pugixml.hpp: No such file or directory is encountered, halting the compilation process.

**Additional Information:**
The temporary solution of copying the pugixml source code to /src/HLS directory suggests that the framework may not be properly included or configured in the build process. Further guidance on resolving this issue would be appreciated.

Ansaya, 2024-04-17:
Hi, you have not correctly initialized the submodules in the repository. After cloning the repo, launch the following: `git submodule update --init`, which should solve your issue.
Further, the `--enable-debug` option will perform a fully unoptimized build, which results in a very slow compilation and execution time. I suggest you use `--disable-release` instead, which enables debug prints and assertions without affecting execution speed too much.

zhaiyunfan, 2024-04-17:
It works, thank you for your help.
In this case, I use the debug option because I want to make some contribution to the resource-sharing feature. I will try release.

By the way, I am trying to modify the source code but the development document is incomplete, where can I find more documents about development?

Ansaya, 2024-04-24:
There is not much documentation for development; it is still a work in progress. I can guide you through if you could share some information on what you'd like to implement.

zhaiyunfan, 2024-04-25:
Thank you very much! We have an idea about resource sharing, aiming to optimize the issue of resource sharing between different layers when using HLS for neural networks. Since the original C++ code is designed using a factory pattern, I encountered some troubles while reading the code.

I would like to know which module or data structure I should focus on if I want to optimize the resource sharing algorithm. Currently, I am puzzled about how your HLS framework identifies shareable substructures from the graph structure.