ferrandi/PandA-bambu #319: Fail to build in docker container

sheldonz7, 2024-04-24:
Hi, I'm trying to build Bambu in a docker container with ubuntu20.04 image.
Followed all the installation instructions, was able to do the configure, but always fail the build with errors below:

`/usr/bin/ld: /workspace/PandA-bambu/obj/ext/sollya/.libs/libsollya.a(external.o): in function `externalPlot':
external.c:(.text+0x50b): warning: Using 'dlopen' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(nanohttp.o): in function `xmlNanoHTTPConnectHost':
(.text+0xb3b): warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/bin/ld: (.text+0x9ba): warning: Using 'gethostbyname' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(xzlib.o): in function `xz_decomp':
(.text+0x7be): undefined reference to `lzma_code'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(xzlib.o): in function `xz_head':
(.text+0xbc0): undefined reference to `lzma_auto_decoder'
/usr/bin/ld: (.text+0xcc2): undefined reference to `lzma_properties_decode'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(xzlib.o): in function `__libxml2_xzclose':
(.text+0x1cb5): undefined reference to `lzma_end'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(encoding.o): in function `openIcuConverter':
(.text+0xfd0): undefined reference to `ucnv_open_66'
/usr/bin/ld: (.text+0xffc): undefined reference to `UCNV_FROM_U_CALLBACK_STOP_66'
/usr/bin/ld: (.text+0x1001): undefined reference to `ucnv_setFromUCallBack_66'
/usr/bin/ld: (.text+0x1020): undefined reference to `ucnv_open_66'
/usr/bin/ld: (.text+0x103b): undefined reference to `ucnv_close_66'
/usr/bin/ld: (.text+0x1073): undefined reference to `UCNV_TO_U_CALLBACK_STOP_66'
/usr/bin/ld: (.text+0x1078): undefined reference to `ucnv_setToUCallBack_66'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(encoding.o): in function `xmlUconvWrapper':
(.text+0x1208): undefined reference to `ucnv_convertEx_66'
/usr/bin/ld: (.text+0x1292): undefined reference to `ucnv_convertEx_66'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(encoding.o): in function `xmlFindCharEncodingHandler.part.0':
(.text+0x25af): undefined reference to `ucnv_close_66'
/usr/bin/ld: (.text+0x25b8): undefined reference to `ucnv_close_66'
/usr/bin/ld: (.text+0x2675): undefined reference to `ucnv_close_66'
/usr/bin/ld: (.text+0x267e): undefined reference to `ucnv_close_66'
/usr/bin/ld: (.text+0x26c9): undefined reference to `ucnv_close_66'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libxml2.a(encoding.o):(.text+0x26d2): more undefined references to `ucnv_close_66' follow
collect2: error: ld returned 1 exit status`

the problem seems to be some linking issues when building sollya? I'm not really sure, please help! Thanks!

Ansaya, 2024-04-24:
Hi, it seems that you are missing some dependencies. Which installation instructions have you followed? Can you share the exact steps you are following so I can reproduce the issue?
Just to be sure, check that you have installed all the packages listed in this [Dockerfile](https://github.com/ferrandi/PandA-bambu/blob/main/etc/containers/Dockerfile.dev).

sheldonz7, 2024-04-25:
Hi, thank you for your prompt response!
I followed the installation instructions for ubuntu 18.04 or newer from the website (https://panda.deib.polimi.it/?page_id=88),
I first installed all the packages listed there,
Then I used the following commands:
```$../configure --enable-flopoco --enable-opt --enable-glpk --enable-release --prefix=/opt/panda
$make -j16
```

sheldonz7, 2024-04-25:
I think after installing some extra dependencies from the Dockerfile.dev, I was able to get through the compilation and building process, will need to test it to see if it is actually built successfully.

sheldonz7, 2024-04-26:
Hi, later on, I tried to install PandA-bambu with clang11, but after installation, when I tried to use it to compile source code, I always get "error -> Unknown compiler I386_CLANG11".

The command I used to configure the project is:
`../configure --enable-flopoco --enable-opt --enable-glpk --enable-release --prefix=/opt/panda --with clang11=/usr/local/bin/clang-11`

Did I miss any step or make a mistake somewhere during the installation?

Ansaya, 2024-04-26:
The option you are using to specify the compiler should only be used when you want to point Bambu to a standalone distribution of clang or gcc compiler (like one from the tar you can download from LLVM project on github). If the compiler is installed system wide the configure script should be able to find it without any hint.
You should be able to tell if the compiler has been found and it is successfully recognised by looking at the configure log. It will say "OK, we have found a compiler" at some point after specific checks are performed.
Finally, check that you have installed both clang-11 and libclang-11-dev packages from APT.

Ansaya, 2024-04-26:
You may also find something useful in the doxygen documentation. Here is the web version https://docs.bambuhls.eu/.

sheldonz7, 2024-04-28:
hi Michele,
I used clang-11 built from source located in /usr/local/bin, which I think it failed to pick up. It works when I install clang from apt. Thanks!

closed 2024-05-09 by sheldonz7