ferrandi/PandA-bambu #326: Compile Bambu with --debug parameter

sheldonz7, 2024-05-27:
Hi, I was trying to print those debugging messages in Bambu to investigate its execution flow, but even after I configured the project with --enable-debug flag, the --debug runtime parameter still doesn't appear in Bambu. When I pass this parameter to Bambu program, e.g.,
```
bambu atax.c --top-fname=atax --pretty-print=out.c --print-dot --compiler=I386_CLANG13 --debug
```
it throws error "Bad Parameters format", as it should since --debug parameter is not supported. How should I do to enable this parameter?

Thanks.

Ansaya, 2024-05-27:
Hi,
I have a build from the dev/panda branch configured with --disable-release, and passing the `--debug` parameter works fine. These kinds of options have not been modified in a long time. Thus, I am expecting them to work the same on any branch.
What branch are you compiling? Could you also share the full configure command line?

sheldonz7, 2024-05-28:
hi,
oops, I didn't put the --disable-release flag when configure the project, thought it won't conflict with debug. After putting that flag in it works fine on the dev/panda branch. Thanks a lot!

Cheers

closed 2024-06-04 by Ansaya