ferrandi/PandA-bambu #340: Multiple Source Files as Input to Bambu

stefanpie, 2024-09-19:
Does Bambu support passing in multiple source files as well as a single defined top function for high level synthesis? It appears from the docs I have read here [https://panda.dei.polimi.it/wp-content/uploads/ProductiveHLS-1.pdf] and here [https://panda.deib.polimi.it/?page_id=916] only ever indicate that you can pass in a single source file.

The motivation for asking is that I have some HLS designs I would like to benchmark with Bambu that are split up into multiple source files, generally to manage the complexity of a large design.

SerenaC94, 2024-09-20:
Yes, passing multiple files is supported. Have a look at the parallel queries example: [bambu.sh](https://github.com/ferrandi/PandA-bambu/blob/main/examples/parallel_queries/bambu.sh) and you will see various .c files passed as arguments to the command line.

Be careful with following tutorial material from 2022 though, some options have changed. There are more recent tutorials in the repository.

stefanpie, 2024-09-26:
Awesome, thank you for the clarification.

closed 2024-09-26 by stefanpie