ferrandi/PandA-bambu #343: Timing issue on Xilinx backend and sdc scheduling issue

sheldonz7, 2024-11-14:
Dear Bambu team,
When using RTL designs generated by Bambu, I constantly get very tight timing after Vivado implementaion(post route timing report), much worse than what Bambu estimated during its backend run.

With a clock period constraint of 10ns, this is one sample result i get: 
cprf = 1        Bambu estimate:  9.836 ns       Vivado implementation report: 10.469 ns
cprf = 0.7      Bambu estimate:  5.275 ns      Vivado implementation report:  9.604 ns  

I also tried to switch to speculative sdc scheduling by including the -s option, but i receive errors, as provided in this file:
[stderr.txt](https://github.com/user-attachments/files/17745596/stderr.txt)

For these experiments, this is the command i use:
```
bambu --top-fname=k2mm --print-dot --compiler=I386_CLANG13 -O2 --debug 4 --verbosity 4 --device=xcku060-3ffva1156-VVD --clock-period=10 --disable-function-proxy --generate-interface=INFER -s ../k2mm.c
```

this is the C code I used:
[k2mm.c.txt](https://github.com/user-attachments/files/17745689/k2mm.c.txt)
[k2mm.h.txt](https://github.com/user-attachments/files/17745685/k2mm.h.txt)

fabrizioferrandi, 2024-11-15:
A first quick comment. --debug 4 is not working very well. Please use --debug-classes instead. This option raises the debug level for a specific calls/step. An example is --debug-classes= parametric_list_based,cdfc_module_binding, where the debug level is raised to the maximum verbosity for the list-based scheduling and the module binding steps.

fabrizioferrandi, 2024-11-15:
Regarding the -s option, it matters only if you have if statements since it activates the code motion/speculation sdc-based scheduling.

sheldonz7, 2024-11-20:
Hi, thank you for the clarification! Regarding resource sharing, I'm already using --disable-function-proxy which should permit more than 1 functional units for operations like fadd right? are you saying using -C to enforce the number of specific functional units being generated in the design rather than relying on module binding (which I believe generates the minimum number of functional units that is necessary).

Ansaya, 2024-11-20:
Hi, --disable-function-proxy and -C are two separate options not to be confused. 

Function proxies are used to share a hardware function between multiple callers. As an example, say you have three functions A, B, and C, where A and B both call C: with function proxies enabled, the tool will generate a single instance of C and use it for both A and B, while using --disable-function-proxy will result in two dedicated instances of C, one for A and one for B.

Setting the number of functional units for a given operation means that you are allowing the module binding to use a given number of functional units instead of a single one.