ferrandi/PandA-bambu #344: Unable to build Bambu 0.97 on Ubuntu 22.04

martinwz, 2024-11-18:
Hi, 
Recently, I have been trying to build Bambu 0.97 on my server, but every time I make it, there is an error, as shown below. The os of the sever is Ubuntu 22.04 and I have installed all required packages by using the command in the INSTALL. I have tried many times, but have not been successful. This is troubling me, may I ask where the problem lies?
[image]

SerenaC94, 2024-11-18:
Hi,
are you following the INSTALL file from the 0.9.7 version? The configuration procedure has changed recently.

martinwz, 2024-11-18:
Yes, I download the .zip file of Bambu 0.9.7 and followed the INSTALL in the .zip file. So, where the problem lies?

Ansaya, 2024-11-18:
Hello,
The error is due to a missing if guard and triggers when no valid Clang/LLVM compiler is found on the build machine. It has been solved in newer releases. I suggest you avoid using 0.9.7 since it is very old and not supported anymore (as well as all releases that are not the latest).

closed 2024-12-18 by Ansaya