ferrandi/PandA-bambu #345: Why does Bambu 24.3 keep reporting errors

martinwz, 2024-11-18:
Hi,
I compiled Bambu 24.03 from the source code. When I use Bambu 24.03 to synthesize code, it keeps reporting the following errors. Could you please help me identify where the problem might be? My OS is Ubuntu 22.04.
[image]

SerenaC94, 2024-11-18:
Hi,
can you share the command-line you used? Which frontend compiler did you select?

SerenaC94, 2024-11-18:
Try running it with --compiler=I386_CLANG12 -v4

Ansaya, 2024-11-18:
Hello,
Compiling Bambu is not really a straightforward procedure, I am sorry for that. I will need some more information to give you a proper answer. What is the configure command line you used for the project configuration? What compilers do you have installed on your system and what version?

Bambu is tested with specific versions of each frontend compiler it supports and needs them to be available and working on the machine for the build process to succeed. Based on the warning message that I see in your log, it seems that you do not have Clang installed at all (or at least Bambu was not able to detect it or it was not happy with the distribution). You may download proper compiler version from https://release.bambuhls.eu/compiler/ as well as pre-compiled AppImage from https://release.bambuhls.eu/ (which I strongly suggest if you are not planning on modifying the codebase).

Ansaya, 2024-11-18:
You may find the 2024.03 release here https://release.bambuhls.eu/bambu-2024.03.AppImage. You may use the [2024.06-dev](https://release.bambuhls.eu/bambu-2024.06-dev.AppImage) release as well, it is not a proper release, but is stable as well.

carherlu, 2024-12-18:
Hi @Ansaya it seems the Bambu AppImages are not currently accessible? It this a sever problem or have you remove them and native instalation is now the only way to use it?. Thanks

Ansaya, 2024-12-18:
Hi @carherlu, unfortunately, our server infrastructure is under maintenance. I hope this ends by the end of the day, but I have yet to be given information on how long it will take. I am sorry for that.

closed 2024-12-18 by Ansaya

martinwz, 2025-04-25:
May I ask when you can release Bambu-24.10.AppImage?