ferrandi/PandA-bambu #346: Appimage Binary File Missing

ZafirSadik, 2025-01-03:
I tried to use the google colab notebooks from this repo. It doesn't work. I also tried to install Bambu in different other methods but the Appimage Binary file is missing. That's why I cannot install Bambu.
Am I doing something wrong? Can anyone guide me on a proper way to install Bambu?

Ansaya, 2025-01-07:
Hello, servers hosting the Bambu HLS AppImage web page have been under maintenance. They should be up and running now. I'm sorry for the inconvenience.

closed 2025-01-08 by ZafirSadik

martinwz, 2025-04-11:
> Hello, servers hosting the Bambu HLS AppImage web page have been under maintenance. They should be up and running now. I'm sorry for the inconvenience.

Hi, may I ask if bambu-2024.10 appimage is available now? I am troubled by building Bambu-2024.10 from source code.