ferrandi/PandA-bambu #351: Cannot build bambu-24.10 on Ubuntu 22.04.4

martinwz, 2025-05-10:
Hi,
I am pleased that you have released a new version of Bambu (Bambu-24.10). However, when I try to build it on my PC (os: Ubuntu 22.04.4), I failed after running the command 'make'.  It reported that 'No rule to make target abseil-cpp/absl/base/internal/cycleclock.cc,  needed by “abseil-cpp/absl/base/internal/libabseil_la-cycleclock.lo” '. So, how to solve this problem?

Best wishes,
Martin

Ansaya, 2025-05-11:
Hi, have you updated the git submodules? You can do that using the `git submodule update --init` command line.

martinwz, 2025-05-11:
> Hi, have you updated the git submodules? You can do that using the `git submodule update --init` command line.

Thank you very much! I can build Bambu on my PC. However, when I use Bambu 24.10 to synthesize my code, it report 'Front-end compiler return an error during compilation 21'. I install gcc-7.5 and clang-12 on my PC. So, where did it go wrong?

Ansaya, 2025-05-11:
In case of error, you should increase verbosity to see the underlying error. You can do that by passing `-v4` to the bambu command line. Also, you may use the `--no-clean` option, so that the tool will not remove the panda-temp folder from the compilation directory and you can inspect the scripts that are generated and executed at runtime.

martinwz, 2025-05-12:
> In case of error, you should increase verbosity to see the underlying error. You can do that by passing `-v4` to the bambu command line. Also, you may use the `--no-clean` option, so that the tool [...]

Thanks! I tried the `--no-clean` option. In the __gcc_output file, it reported '/usr/include/bits/stdio2.h:111:1: error: expected identifier or ‘(’ before ‘{’ token'. Is this error caused by building Bambu incorrectly?

agostini01, 2025-05-12:
> Hi, have you updated the git submodules? You can do that using the `git submodule update --init` command line.

I had the same error:

```
No rule to make target abseil-cpp/absl/base/internal/cycleclock.cc, needed by
```

And updating the submodules resolved my issue.
Thank you!

martinwz, 2025-05-13:
> > Hi, have you updated the git submodules? You can do that using the `git submodule update --init` command line.
> 
> I had the same error:
> 
> ```
> No rule to make target abseil-cpp/absl/base/internal/cycleclock.cc, needed by
> ```
> 
> And updating the submodules resolved my issue. Thank you!

I also resolved my issue. However, when I use Bambu 24.10 to synthesize my code, it report 'Front-end compiler return an error during compilation 21'.  I tried the --no-clean option. In the __gcc_output file, it reported '/usr/include/bits/stdio2.h:111:1: error: expected identifier or ‘(’ before ‘{’ token'. May I ask what is your environment?