ferrandi/PandA-bambu #361: Bambu 24.10 crashes

martinwz, 2025-07-26:
- **Tool Version:** https://release.bambuhls.eu/bambu-2024.10.AppImage
- **OS Version:** Ubuntu 22.04.4 LTS 
- **Frontend Compiler Version:** clang-16
- **Simulator:**Iverilog 11.0
**Command Line Used:** /media/oscar/d/wz/download/Bambu/bambu-2024.10.AppImage -v 4 --top-fname=result tp.c >bambu.log 2>&1;

**Description of the Bug:**
Bambu 24.10 crashes and cannot generate verilog code. It reports segmentation fault.  

**Steps to Reproduce:**
Run ./run_syn.sh. (Please modify the path of bambu 24.10 in the shell file according to your environment)

**Expected Behavior:**
You will find the detailed crash information in bambu.log. 

**Actual Behavior:**
None.

**Attachments:**
All materials required for reproducing the problem in the attachment.
[crash_1_4.zip](https://github.com/user-attachments/files/21444887/crash_1_4.zip)

**Additional Information:**
None.

fabrizioferrandi, 2025-07-29:
I was able to reproduce the problem. It comes when a phi becomes something like a self-pointing phi:
```
define i32 @Result() local_unnamed_addr #0 {
br label %2
2: ; preds = %2, %0
%3 = phi i32 [ 0, %0 ], [ %3, %2 ]
.....
```

Besides that issue you found, I've reviewed the submitted code and I've some concerns. Using **Csmith** to test an HLS compiler is likely not very effective. The generated code includes many constructs that are typically unsupported by HLS tools.
For example, the following snippet is unlikely to be synthesizable by any HLS tool I'm familiar with:
```

static int32_t *g_15 = &g_16[6][1][3];
static int32_t ** volatile g_35 = &g_15;

```
While extending Bambu to support this kind of memory initialization would be technically possible, doing so would serve little practical purpose. In my experience, such global variable initializations are extremely rare in real-world applications.

martinwz, 2025-08-20:
> I was able to reproduce the problem. It comes when a phi becomes something like a self-pointing phi:
>
> ```
> define i32 @Result() local_unnamed_addr #0 {
> br label %2
> 2: ; preds = %2, %0
> %3 = phi i32 [ [...]

closed 2025-08-20 by martinwz